Mechanical-Advantage/AdvantageKit #6: Publish to a different package registry

jonahsnider, 2022-02-27:
Requiring users to setup an access token for GitHub Package Registry just to install the AdvantageKit libraries is not very user-friendly. Could binaries be published to a different registry that makes installation easier?

camearle20, 2022-02-27:
This is unfortunately not a requirement we can control, GH packages does not allow any anonymous access for maven targets for some reason.  We are providing an API key that can be used publicly (it just changed yesterday, so please be sure you update it in your build file from the instructions).

Do you have any suggestions for other free maven package hosting options?  My go-to was Bintray, but that got shut down recently.

jonahsnider, 2022-02-27:
Here's some free alternatives I've seen others in the FRC ecosystem using:
- [JitPack](https://jitpack.io/) - [Oblog](https://github.com/Oblarg/Oblog) uses this. GitHub has [documentation](https://docs.github.com/en/actions/publishing-packages/publishing-java-packages-with-maven#publishing-packages-to-the-maven-central-repository) on how to publish here with GitHub Actions.
- [The Maven repository](https://mvnrepository.com/) - Used by Kauai Labs, WPILib, and a few FRC teams.

camearle20, 2022-02-27:
We thought about JitPack, but unfortunately we're building with Bazel and not one of the supported build tools, plus JitPack can cause unexpected timeouts the first time a new release is pulled since it builds it on the fly.  mvnrepository is not a package hosting service, but rather just a search tool for indexing a bunch of different hosted maven repositories.  We could try to host on maven central, but I'm not sure our library matches their requirements.

For now, I think we have a working solution with GitHub Packages.  We provide an API key for downloading which doesn't require any additional setup by users; I think your build may have failed because you included our old (non obfuscated) API key in your build.gradle and left out our trick of concatenating two strings together, causing GitHub to flag this and nuke our API key for both you and everyone else using AdvantageKit.  We've documented this more clearly in our setup instructions, but that was the reason we were appending two strings for seemingly no reason.

We are actively investigating other options for hosting, but for now I think we're sticking with packages.  Hopefully GitHub will fix this issue sometime this year so we don't have to have users paste the key into their build file anymore.

jonahsnider, 2022-02-28:
> We thought about JitPack, but unfortunately we're building with Bazel and not one of the supported build tools, plus JitPack can cause unexpected timeouts the first time a new release is pulled [...]

Makes sense.

> For now, I think we have a working solution with GitHub Packages. We provide an API key for downloading which doesn't require any additional setup by users; I think your build may have failed [...]

I suspected that concatenating the strings was to avoid some kind of automated scanner, but didn't realize GitHub would automatically revoke the key. Sorry about that!

> We are actively investigating other options for hosting, but for now I think we're sticking with packages. Hopefully GitHub will fix this issue sometime this year so we don't have to have users paste the key into their build file anymore.

I would be pleasantly surprised if GitHub did remove that requirement, here's hoping.

closed 2022-02-28 by jonahsnider

camearle20, 2022-02-28:
> Sorry about that!

No worries!  We figured this would happen at some point, and it's our bad for not documenting it more clearly.

To add some closure to this, we are looking more seriously now at just publishing to Maven Central.  This will most likely be something we do in the offseason.