Mechanical-Advantage/AdvantageKit #8: Should each hardware IO class get its own sim class?

jonahsnider, 2022-03-14:
After a very, very, VERY lengthy discussion on the FRC Discord it seems like the simulation code that teams write should be hardware specific, which doesn't match the pattern that Advantage Kit uses.

For example, suppose you have an arm subsystem with `ArmIO`, `ArmIOFalcon500`, and `ArmIOSparkMax` classes.
If you want to create an `ArmIOSim` class you will probably use WPILib's `SingleJointedArmSim` class, which means you need to provide a `DCMotor` instance.
You could do `DCMotor.getFalcon500(1)` or `DCMotor.getNEO(1)`. How do you decide which one to use?

It seems like in this situation you'd need a `ArmIOFalcon500Sim` and a `ArmIOSparkMaxSim` class which would share a lot of the same code but with minor differences in how the simulation hardware is setup.

jonahb55, 2022-03-14:
I read through most of the discussion on Discord, let me know if I'm missing some context.

The primary purpose of the IO layer is to disable vendor-specific hardware when replaying from a log file; we suggest using an anonymous inner class to accomplish this. Support for multiple hardware implementations (e.g. Falcon500 vs. SparkMax), or simulated implementations, is mostly a nice side effect of that system. The IO layer is a structure that we've personally found to work well, but it is by no means required to use AdvantageKit. From the "Getting Started: Logging" guide:

> Any structure will work where all input data flows through an inputs object implementing `LoggableInputs` and the two methods `fromLog` and `toLog`. Feel free to make use of whatever structure best fits your own requirements.

Depending on how each team chooses to implement (or not implement) simulation, they can adapt the structure to their own needs. For example, this could include creating IO implementations that inherit from each other or take extra constructor arguments. AdvantageKit is set up to be flexible.

With that said, I think that a single `ArmIOSim` class can still make sense. Based on my understanding, WPILib and vendor libraries provide two key tools for simulation:

1. Physics simulation classes like `SingleJointedArmSim` or `DifferentialDriveSim`, which model the physics of a mechanism without any vendor-specific logic.
2. Hardware specific simulation classes like `PWMSim` or `REVPhysicsSim`. While they may offer some basic physics simulation support, they primarily serve as an interface to inject data to the equivalent `PWM` or `CANSparkMAX` objects.

The intended workflow seems to be using these classes in a `simulationPeriodic()` method. The first set of classes are used to simulate the mechanism, then the results can be injected into the normal user code using the second set of classes. I haven't used a system like this myself, but that's my best understanding of how it's meant to work.

Using an IO layer negates the purpose of the second set of hardware specific classes. The way our team uses it, the abstraction in the IO layer *already removes any vendor-specific code.* Data is injected by using a different IO implementation - there's no need for vendor classes to deal with this. This means that the first set of classes can be used to provide the simulated implementation. Calls like `DCMotor.getFalcon500(1)` aren't doing anything vendor specific beyond grabbing a set of constants to roughly match the physics of a Falcon. This means that the only purpose of separate `ArmIOFalcon500Sim` and `ArmIOSparkMaxSim` classes would be if you really needed to simulate the physics of a specific motor. Generally, we choose to treat our simulated robot as an entirely separate robot with a different set of constants. This means that there's no point to having multiple sim implementations. The sim robot might "have two Falcons," but that no more important to the rest of the code than "the competition bot has two NEOs" or "the practice bot has one Falcon."

In short, I don't think there are any pressing changes required to AdvantageKit right now. Teams are able to structure simulated code in whatever way makes sense to them.

jonahsnider, 2022-03-14:
> Generally, we choose to treat our simulated robot as an entirely separate robot with a different set of constants. This means that there's no point to having multiple sim implementations.

Thanks for clarifying this. This is super super helpful!

> In short, I don't think there are any pressing changes required to AdvantageKit right now. Teams are able to structure simulated code in whatever way makes sense to them.

I agree. Thanks for the info!

closed 2022-03-14 by jonahsnider