Mechanical-Advantage/AdvantageKit #9: CAN: Message not found org.littleton...... (LoggedRobot.java:49)

jonahsnider, 2022-03-25:
We (581, 6036) have noticed a constant stream of the following error (warning?) to the console when running robot code using Advantage Kit:

[image]

Unfortunately I don't have any better logs to use in this report, but I can save some this Saturday at our next meeting.
This is behavior that's been going on for quite a while now, not a recently introduced issue.

The relevant code appears to be related to PDP information. In case this is helpful: 581 uses a classic PDP, 6036 uses a REV PDH.

jonahb55, 2022-03-25:
We're heading to an event this weekend, so unfortunately I won't have time to properly investigate this until next week. You're correct that it's related to saving PDP information (specifically the set of default inputs defined in [`LoggedSystemStats`](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/junction/core/src/org/littletonrobotics/junction/inputs/LoggedSystemStats.java)). It defaults to reading from the CTRE PDP though, and you can use a call [like this](https://github.com/Mechanical-Advantage/RobotCode2022/blob/a376aaa45ab5a862468cadd541a978e51d0cda00/src/main/java/frc/robot/Robot.java#L97-L98) to override the module type and CAN ID. I'm not sure that's the whole issue if it's happening on the PDP too, but it may be part of the problem if you aren't setting the type explicitly.

jonahb55, 2022-03-25:
It would also be good to confirm whether the logged current data is accurate (under "/SystemStats/PowerDistributionCurrents").

jonahsnider, 2022-03-25:
> We're heading to an event this weekend, so unfortunately I won't have time to properly investigate this until next week.

Good luck at the event! This doesn't appear to be a major issue, so working on it in the future sounds fine. The only real impact is the possible performance cost of constantly writing big stacktraces to stderr.

> I'm not sure that's the whole issue if it's happening on the PDP too, but it may be part of the problem if you aren't setting the type explicitly.

I'll try manually setting this and seeing if it helps.

> It would also be good to confirm whether the logged current data is accurate (under "/SystemStats/PowerDistributionCurrents").

It's 0-23 elements that are all `0`.

jonahb55, 2022-03-29:
Did manually setting the power distribution type help on either robot? I noticed that in the log you posted in #13, reading system stats data is taking much longer than it should. This makes me suspect that there's some sort of timeout involved where it fails to read the data. The top graph is from your log and the bottom is from one of ours

[image]

[image]

jonahsnider, 2022-03-30:
> Did manually setting the power distribution type help on either robot?

Yes, it fixed the error in the console. The logged values are still all zeroes though.

> This makes me suspect that there's some sort of timeout involved where it fails to read the data.

This seems reasonable. I'm doing a more in depth investigation into our performance issues. It's possible this is related, not sure yet.

jonahsnider, 2022-04-01:
I'll also add that for whatever reason our PDP was using port 19 or something instead of whatever the WPILib default is.
Keeping the existing behavior seems fine I guess. Perhaps it should be opt-in though.

jonahb55, 2022-04-01:
I'm glad you got it resolved. Adding support for the PDH and multiple CAN IDs was a bit of a last minute addition, but we can reevaluate the default behavior in the future.

closed 2022-04-01 by jonahb55