Mechanical-Advantage/AdvantageKit #11: Selectively streaming log entries

jonahsnider, 2022-03-28:
Is there a way to select which entries should be streamed over HTTP via `LogSocketServer`?

We're working to lower our robot's loop time and suspect that streaming tons of sensor data to the driver station during a match is probably not great. Ideally we'd save everything to the USB and stream odometry to the driver station for display in the field visualization.

jonahsnider, 2022-03-28:
Ah, I had assumed the odometry feature in Advantage Kit was meant for both debugging & during matches. Using Shuffleboard works, we'll use that for now. Although I do much prefer the Advantage Scope odometry view... the WPILib stuff always seems to use the crustiest background images they can find.

itsmichaelyu, 2022-12-30:
Since AdvantageKit now uses NT4, is this still a problem? I'm not sure how exactly it uses NT4, but I would assume since all the data is already being streamed, it theoretically shouldn't have an impact?

jwbonner, 2022-12-30:
AdvantageKit publishes every to the NT server (no network usage because it's also on the RIO), but NT4 now uses a pub/sub system so only the fields that each client subscribes to are sent. Theoretically that would mean that (for example) only the displayed fields on a dashboard are sent. Note that AdvantageScope actually subscribes to all fields because it's often useful to scroll back in an NT log and look at different fields.

I don't think there's a compelling use case for being more selective in how AdvantageKit publishes data. If loop times or bandwidth are a concern, you can always publish dashboard data from the user code and disable the `NT4Publisher` when running on the field. I'll probably close this issue in a few days if there are no objections.

jwbonner, 2023-12-29:
I said I would close the issue in a few days, and it's now been a year 😆. I don't think anything more is required on the AdvantageKit side after the switch to NT4.

closed 2023-12-29 by jwbonner