Mechanical-Advantage/AdvantageKit #13: Broken timestamps from replay

jonahsnider, 2022-03-29:
[image]

This is a screenshot of a log I recorded using the replay feature in a simulator on my PC.
As you can see, the timestamps start off pretty norma, then display an incredibly large number. This seems to cause all subsequent timestamp data to become unreliable.

Here is the log file in question:
[Log_22-03-26_14-04-39_sim.rlog](https://github.com/Mechanical-Advantage/AdvantageKit/files/8368845/Log_22-03-26_14-04-39_sim.rlog.txt)

I *think* I've observed this behavior with the huge timestamps on actual hardware before but can't be sure.

I suspect that some corruption is occurring in the RLOG file which causes the "cursor" reading through the file to become messed up, which results in subsequent log entries being corrupt.

jonahb55, 2022-03-29:
Do you have the original log file from the RIO to post as well? Presumably it doesn't exhibit the same issue, but I'm curious if there's someone odd that might cause problems when the simulator re-encodes it. Does this happen consistently when replaying this log?

jonahsnider, 2022-03-30:
> Do you have the original log file from the RIO to post as well? Presumably it doesn't exhibit the same issue

Surprise! This issue occurs on actual hardware, and the replay doesn't actually seem to cause any issues.

Here's a fresh RLOG for you to use. This was recorded on actual hardware.
[Log_22-03-29_19-26-05.rlog.txt](https://github.com/Mechanical-Advantage/AdvantageKit/files/8377429/Log_22-03-29_19-26-05.rlog.txt)

Normally in a situation like this I'd refer to the logs to help track down the issue, but given that those are unreliable... I'm not sure what the best course of action is here.

---

I am still not sure what causes our loop times to be so slow. We're running a roboRIO v1 and logging a lot of values, perhaps that's related? I tried running an actual JVM profiler but had issues and ran out of time at today's meeting so I can still only guess where the performance issues come from.

Maybe this is a race condition where two log entries are processed in parallel, one overwriting the other and causing corruption.

Maybe the issue comes from WPILib's tick catchup feature when your robot loop overruns. You can read about this behavior [here](https://www.chiefdelphi.com/t/what-actually-happens-if-your-commands-takes-more-that-20ms/406971?u=jonahsnider). Perhaps the time difference between two calls to record logs is a tiny number and somehow is used in a `x / tiny_duration` which outputs the comically large timestamp (fun fact! the logs indicate that our robot is capable of time traveling 14 Gyr (gigayears) into the future)

---

I wonder, are you guys able to reproduce this if you try forcing loop overruns? Either by doing something stupid in your `robotPeriodic()` or by setting the loop period to be a tiny number like `2ms`.

I'll continue trying to track down our performance issues tomorrow, provided I can get the profiler working.

camearle20, 2022-03-30:
> Perhaps the time difference between two calls to record logs is a tiny number and somehow is used in a x / tiny_duration which outputs the comically large timestamp

I don't think we're using dt anywhere within the logger.  The timestamp that is written to the logs is coming straight from the underlying HAL call to read the FPGA timestamp.  There's a unit conversion in there to go from the 32 bit integer microseconds value to a double in seconds, so potentially if something is getting messed up with the upper byte(s) of that number odd results could come out, but I don't suspect that is what is happening here.

To my knowledge we haven't seen this happening in any of our logs.  We do have occasional loop overruns, but still haven't seen this behavior.  I can poke around some of the core code in conduit to see if we're missing something when handling the FPGA timestamps, but beyond that I don't think we're manipulating the data coming from WPILib in any way.

jonahsnider, 2022-03-30:
In that case, I suspect this is probably not related specifically to the timestamps but is more likely a broader log corruption issue. The race condition idea I mentioned earlier seems semi-reasonable, but you know more about how that code functions.

jonahb55, 2022-03-31:
I'm definitely curious if that made a difference to the log file. I took a look at the one you posted in a hex editor, and the cycle around ~47 seconds does look corrupted (though it looks fine afterwards). I checked through the code that handles RLOG encoding, and I don't see any way it could have produced that data. It might also be worth trying different USB sticks to see if one might be faulty.

jonahsnider, 2022-03-31:
> I can take a look at the log files later tonight and see if they are still corrupted, or if that was the fix.

Still corrupted :(
[RLOG from today with the performance improvements applied](https://github.com/Mechanical-Advantage/AdvantageKit/files/8386160/Log_22-03-30_19-16-11.rlog.txt).

The logs *look* fine when streamed over the log server. I'm like, mostly confident there's nothing wrong with the streamed logs.

---

> It might also be worth trying different USB sticks to see if one might be faulty.

This might be it. I can try running the robot with a different model USB.
We're using [these tiny Sandisk USBs](https://www.amazon.com/dp/B09FFK1QLR). I bought them off Amazon recently, haven't tried using them for anything else.

In the past I've heard borderline conspiracy theories from other FRC teams sharing super low level USB IO issues they've experienced, hopefully this isn't something like that.

jonahsnider, 2022-03-31:
> The RLOG data is only encoded once by AdvantageKit - the same data is saved to the file and sent over the network. If the streamed data is fine, this definitely looks like some sort of issue with the USB stick.

Hopefully this is the case, replacing a USB drive is pretty easy. I can confirm whether or not streamed logs suffer the same corruption issue today.

Tyler from WPILib said "there's been lots of reports of filesystem corruption this year in general", not sure what specifically that means but it seems bad. He later said "the filesystem *is* one of the only things that changed this year".

> That might involve adding a sanity check to the timestamp in Advantage Scope or including a checksum at the start of each cycle.

I agree. This seems like an essential feature to add.
Here's some relevant reading to help with implementing an integrity check:
- https://en.wikipedia.org/wiki/Cyclic_redundancy_check
- https://en.wikipedia.org/wiki/Computation_of_cyclic_redundancy_checks
- https://en.wikipedia.org/wiki/File_verification

jonahb55, 2022-03-31:
We use FAT32 on our USB drives. As for repairing files manually, you could look at the RLOG [specification](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/docs/RLOG-SPEC.md) if you haven't already. I've been adding extra console logs to Advantage Scope to print the byte position when it reads the relevant timestamp, then looking through the hex data manually. If there are particular files you need me to repair, I can do my best to fix them up in the meantime.

jonahb55, 2022-03-31:
AdvantageKit was writing log data every cycle, and it relied on the OS to handle syncing the data. I just released [AdvantageKit v1.7.3](https://github.com/Mechanical-Advantage/AdvantageKit/releases/tag/v1.7.3), which now buffers the data with a write & flush every 0.25 seconds. I'm also working on adding a sanity check to the timestamps in Advantage Scope. Hopefully those changes are enough to fix this issue. If not, checksums are still an option (though less optimal, since there would still be data loss).

jonahb55, 2022-03-31:
[Advantage Scope v1.17.2](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v1.17.2) now includes a sanity check for timestamps. Data is rejected if the timestamp is less than the previous cycle or more than 15 seconds greater. It will print out the time and byte address of the corrupted data to the console, then search each byte until it finds the next reasonable timestamp. For the few logs you posted, this seems to do a decent job of making the data "not totally broken."

I haven't added the same check to the replay system yet, let me know if that's something you need working. Unfortunately, the utility of a check like this is probably degraded for replay since losing only a single cycle of data can mess up the whole simulation.

jonahsnider, 2022-03-31:
> We use FAT32 on our USB drives.

Good to know.

> As for repairing files manually, you could look at the RLOG [specification](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/docs/RLOG-SPEC.md) if you haven't already. I've been adding [...]

I've skimmed the spec before but if this issue persists I'll try and read it more closely to get a better understanding of how it works.

> If there are particular files you need me to repair, I can do my best to fix them up in the meantime.

Thanks for the offer! Normally I'd say "yes, please do this!", but I think we just got our localization stuff working (🎉) so I think we're good.

> AdvantageKit was writing log data every cycle, and it relied on the OS to handle syncing the data. I just released [AdvantageKit [...]

> [Advantage Scope v1.17.2](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v1.17.2) now includes a sanity check for timestamps. Data is rejected if the timestamp is less than the previous cycle or more than 15 seconds greater. It will print out the time and byte address of the corrupted data to the console, then search each byte until it finds the next reasonable timestamp. For the few logs you posted, this seems to do a decent job of making the data "not totally broken."
> 
> I haven't added the same check to the replay system yet, let me know if that's something you need working. Unfortunately, the utility of a check like this is probably degraded for replay since losing only a single cycle of data can mess up the whole simulation. Ultimately, the only real solution is figuring out how to save data that isn't corrupted in the first place.

Thanks so much for these updates! I'll test them out later today and let you know how it goes.

jonahsnider, 2022-04-01:
Hey guys, good news: I think I figured out what the bug is.
I'm a bit embarrassed I didn't notice/mention this earlier but......
**All the corrupted logs came from Advantage Scope's "Download Logs" feature.**

I've tested the handful of RLOGs I downloaded using that feature: all corrupted.
The same files direct from the USB? Zero errors, every single file. Totally fine.

---

> [Advantage Scope v1.17.2](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v1.17.2) now includes a sanity check for timestamps.

You should also disallow duplicate timestamps, still seeing those leaking through in corrupted log files.

---

I didn't check if the USB itself was the problem during today's meeting but it seems like maybe that's not needed anymore.

jonahb55, 2022-04-01:
[AdvantageScope v1.17.3](https://github.com/Mechanical-Advantage/AdvantageScope/releases/tag/v1.17.3) adds detection of duplicate timestamps and fixes an issue preventing live connections (since the first timestamp can be greater than 15 seconds in that case).

The "Download Logs" feature uses SFTP to pull the log files (the [`fastGet`](https://github.com/mscdex/ssh2/blob/master/SFTP.md#sftp-methods) method of `ssh2` specifically, meaning Advantage Scope never sees the data directly). Does the same issue occur if you download over SFTP with a different client (command line, another app, etc.)?

jonahsnider, 2022-04-02:
> The "Download Logs" feature uses SFTP to pull the log files (the [`fastGet`](https://github.com/mscdex/ssh2/blob/master/SFTP.md?rgh-link-date=2022-04-01T23%3A04%3A28Z#sftp-methods) method of `ssh2` [...]

I can test this tomorrow.

I'm immediately wary of ssh2. There are a few bug reports of file corruption in both `fastGet` and `fastPut`.
IMO the safest option here is to ditch the `ssh` implementation in favor of a custom `.rlog` static file server added into AdvantageKit.

---

I will also mention that each time I've downloaded files using that Advantage Scope feature the robot program was still writing to the files. Perhaps the file length changing caused ssh2's SFTP implementation to get messed up somehow?

jonahb55, 2022-04-02:
If this continues to be problematic, we can see about other approaches to downloading logs. If another SFTP client works, that seems like a decent workaround for now. It's a little odd that this seems be so persistent though. We've been using the same system heavily through two competitions (including logs actively being written) with no issues.

jonahsnider, 2022-04-02:
I tested a bit with plain SCP and Advantage Scope's download feature and was unable to reproduce the corruption issue.
My theory that something was wrong with Advantage Scope was probably incorrect. I suspect that the several fixes made to AdvantageKit and perhaps the performance improvements to our codebase were what fixed this.

Going to mark the issue as closed since this seems to be fixed, thanks again for all your help with this!

closed 2022-04-02 by jonahsnider