Mechanical-Advantage/AdvantageKit #15: Log files generated in sim are not viewable in AdvantageScope

tervay, 2022-06-19:
Generating a logfile in sim is not viewable in AdvantageScope, due to an invalid revision number:

[image]

----

Reproducible repo: https://github.com/tervay/loggerSwerve2/tree/a3746ff899bf4b868b96259bfbc745e9c0a15fc6

VScode will get angry at stuff in there if you don't have the Lombok extension installed, sorry. Any log file in there should work.

AdvantageScope can connect to the simulator live just fine and works as a workaround. Speaking to @camearle20, it seems that the revision number is just completely missing from the log file.

jonahb55, 2022-06-20:
All of the log files you posted start with a valid timestamp around three seconds, which always occurs at the start of a new cycle. However, they don't include any information defining key IDs, which I would expect to be prominent in the first two cycles. Based on that, it looks like some of the initial cycles are failing to write to the log file (but there aren't any partial cycles being written). The data is buffered and written to the log file every 0.25 seconds ([code here](https://github.com/Mechanical-Advantage/AdvantageKit/blob/591fb8689c43b0fb0cf089a7ac3c01b9fdac101b/junction/core/src/org/littletonrobotics/junction/io/ByteLogReceiver.java#L124-L136)), so my best guess is that the first write is failing (or the first few writes).

I'm not seeing the same issue when I simulate your code, so I wonder if this is affected by the OS/filesystem. I'm running macOS 12.3 on an APFS filesystem. Does the same issue occur if you simulate our robot code? I configured [this branch](https://github.com/Mechanical-Advantage/RobotCode2022/tree/sim-log-test) to use the correct robot and write to a log file.

tervay, 2022-06-21:
I originally got the bug when running solely in Win11. I then cloned my project in WSL2, which is running Ubuntu 20.04, and was unable to reproduce it.

I cloned the `sim-log-test` branch of the 6328 codebase on Win11 (I disabled the `createVersionFile` step because I don't have git set up on windows - my workflow is weird - but that shouldn't affect anything), and was able to reproduce it.

So this seems like a Windows issue, somehow.

jonahb55, 2022-06-21:
Interesting. I'll try setting up the environment in my Windows VM and see if I can reproduce the issue. Hopefully there just needs to be an extra flush somewhere to fix it.

commit 8c74ef6b (2022-06-23): Fix renaming of log files

Uses `Files.move(a, b)` as a platform independent solution instead of `a.renameTo(b)`. Fixes #15
closed 2022-06-23 by jonahb55