Mechanical-Advantage/AdvantageKit #16: Usage analytics

jonahsnider, 2022-08-20:
I'm curious, does GitHub or their package registry give any kind of usage analytics? It might be interesting to see (approximately) how many teams are using AdvantageKit and how it's grown in popularity over time.

jonahb55, 2022-08-20:
GitHub Packages doesn't provided detailed analytics as far as I can tell. [Each main package](https://github.com/orgs/Mechanical-Advantage/packages) has ~350 downloads currently, but that's pretty meaningless because most teams have multiple users (and each user might download the packages multiple times depending on caching). I've found the best way to roughly track usage (beyond who we talk to directly) has been to search for the vendordep UUID on GitHub ([example](https://github.com/search?q=d820cc26-74e3-11ec-90d6-0242ac120003&type=Code)). That produces a list of at least 14 teams currently using AdvantageKit in some respect:

401, 581, 1189, 1561, 2239, 2713, 3176, 3181, 4099, 4176, 5298, 5987, 6014, 6328

Obviously the full list is probably a little longer because of any teams that haven't made their code public on GitHub. I'm also not certain that GitHub's search includes branches other than the default one. While I don't have exact numbers, I know that list has grown significantly throughout the offseason.

jonahsnider, 2022-08-20:
Neat, thanks for the info!

closed 2022-08-20 by jonahsnider