Mechanical-Advantage/AdvantageKit #18: Putting Logged RealOutputs on Shuffleboard Tabs

itsmichaelyu, 2022-10-01:
I think this would be a great feature, basically what I envision is when you log a real output, there is a constructor with a boolean where you can choose to enable Putting the data to a Shuffleboard tab. It would just parse the key like this "Intake/Extended", it would split at the "/" and take the "Intake" as the title of the Shuffleboard tab and "Extended" would be the name of the item on Shuffleboard. I could possibly look in to implementing this, but I'm not to familiar with the lower level of AdvantageKit

jonahb55, 2022-10-01:
Funny you should mention it...

The tentative plan for the AdvantageKit V2 is to support NetworkTables as the recommended way of sending live data (Advantage Scope will support NT4). Everything would be under an AdvantageKit table, like "/AdvantageKit/RealOutputs/Intake/Extended", so you would be free to add it to Shuffleboard (or use the auto-populate option for different tabs, probably).

The Advantage Scope support is already ~90% done, and we're waiting until we can pull in WPILib with NT4 to develop the AdvantageKit side of things. We also need to check the performance of writing to NT before we commit to this approach.

jwbonner, 2022-11-13:
NT4 is now supported by the 2023 beta release of AdvantageKit:

```java
Logger.getInstance().addDataReceiver(new NT4Publisher());
```

closed 2022-11-13 by jwbonner