Mechanical-Advantage/AdvantageKit #19: Add labels for PDP channels

scarmain, 2022-10-21:
Would it be possible to add in a function like setChannelNames(string[]) to the LoggedSystemStats class?  

I honestly don't want to remember or want to keep looking up all the power channel names for every trace...

jonahb55, 2022-10-21:
Could you elaborate on the use case for this? What are the traces that you're trying to interpret? If you're looking at log data for motors, I'd generally recommend that you record the current draw reported by the motor controllers ([example](https://github.com/Mechanical-Advantage/RobotCode2022/blob/5c8d90af38c2239fd545cd22edb662667809f88e/src/main/java/frc/robot/subsystems/climber/ClimberIOSparkMAX.java#L91-L92)) as it's more accurate than the PDP/PDH data.

scarmain, 2022-10-21:
So, what is the point of keeping the PDP/PDH data logged?  If the data is "useless", don't save the 24 channels then.  Effectively now, you are logging the PDP channel and the speed controller value, and double logging the "same" data.  (I have no experience who has the better current sensor)

The use case is if we are already collecting the PDP/PDH data, let me say channel 1 is FL drive motor, instead of making me look it up each channel every trace.

camearle20, 2022-10-24:
I don't think anyone said the data is useless... To your point, we do (or are about to with #17 being merged) support disabling PDP/PDH logging altogether. It's also a future goal of ours to support more automatically logging data from motor controllers, which should help the usability here.

The current use case for us for auto-logging the PDP data is just to eliminate our need to use the driver station log viewer and to be able to see that data on the same time series as our other logged data. We don't support using the data in replay as of right now since it isn't something we have found would be necessary. If you did want to be able to log PDP channels with a convenient name, I'd recommend disabling the auto logging and wrapping the interactions with the channels you care about in an "IO layer" (something we touch on briefly in our documentation and which you can see from our 2022 codebase) so that you can explicitly log it with a label of your choosing and also replay that data (probably useful if you're using that data with a PWM motor controller to do something like a current limit).

It has been our experience that the data from the PDP is somewhat lower resolution than the data from smart motor controllers, and that it has a poor update rate, which is why it is treated as somewhat of an afterthought by us. As for why we can't easily make this feature happen for you, the architecture of our logger is not built in a way that we can add arbitrary labels to already logged fields, and it would require at the very least a new revision of our RLOG format to make this possible. Additionally, as we're planning on moving to the WPILOG format in the near future, it would become essentially impossible. An easier way to do it would be to support custom labels in AdvantageScope. This isn't currently on our roadmap, but I could see about getting something more general like the ability to add "favorite" fields with custom shorthand labels to AdvantageScope.

scarmain, 2022-10-25:
Thanks for your time.  Honestly, this logger took some work to configure, and with your scope supporting WpiLog (I downloaded a 2.0 nightly build and it worked great), we'll probably use their logging solution for next season.  I did like seeing your different way to abstract and replay inputs (it took me like a week to really understand the concept), and we might use some of that in our code this season.

I really like your scope tool (the main reason we tried this logging framework), and we WILL be using it this season a lot.  I have a couple feedback items over there I will leave.  (not about this, with the WpiLib logging, I just have to name my PDP channels when I log them, so no favorite fields are needed)

closed 2022-10-25 by scarmain