Mechanical-Advantage/AdvantageKit #25: wpilibio for macOS arm64?

gcschmit, 2022-12-09:
We have updated last season's code to the 2023 beta and are testing against the latest AdvantageKit beta. (The new set of features look amazing, btw.) When I attempt to simulate the robot code, the following exception is generated:

```
Error at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2670): Unhandled exception: java.lang.UnsatisfiedLinkError: no wpilibio in java.library.path: [/Users/geoff/GitHub/Swerve-Library/build/jni/release]
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2670)
        at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:830)
        at java.base/java.lang.System.loadLibrary(System.java:1873)
        at org.littletonrobotics.conduit.ConduitJni.<clinit>(ConduitJni.java:11)
        at org.littletonrobotics.conduit.ConduitApi.<init>(ConduitApi.java:40)
        at org.littletonrobotics.conduit.ConduitApi.getInstance(ConduitApi.java:28)
        at org.littletonrobotics.junction.Logger.periodicBeforeUser(Logger.java:178)
        at org.littletonrobotics.junction.Logger.start(Logger.java:147)
        at frc.robot.Robot.robotInit(Robot.java:97)
        at org.littletonrobotics.junction.LoggedRobot.startCompetition(LoggedRobot.java:59)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:343)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:413)
        at java.base/java.lang.Thread.run(Thread.java:829)
```

I believe this is because  org.littletonrobotics.akit.conduit.conduit-wpilibio is not built for macOS arm64 since the vendor dependencies AdvantageKit.json doesn't list that as a supported platform.

Are there plans to support macOS arm64? Is it possible for me to build this target locally so we can continue testing?

camearle20, 2022-12-09:
This is on our agenda as something we'd like to get done before this coming season. I have been putting it off for a while though... I'll make a push to try to get at least the ability to run a local build of this on an M1 Mac functional. Getting this working in CI may be slightly more challenging due to how Bazel defines toolchain platforms.

gcschmit, 2022-12-09:
I'm unfamiliar with Bazel, but should I try to get it installed and build locally. Or, will that inevitably be an exercise in frustration?

jwbonner, 2022-12-09:
Cameron can comment on Bazel, but I'll add that for now you also have the option of installing the Intel version of the WPILib beta and using Rosetta (maybe just switching the JDK to an x86 build would work too)? We've been testing this way on our macOS devices.

gcschmit, 2022-12-09:
Clever; I'll try that first.

camearle20, 2022-12-09:
> I'm unfamiliar with Bazel, but should I try to get it installed and build locally. Or, will that inevitably be an exercise in frustration?

The learning curve is unfortunately somewhat steep. We have a script at the root of the repo called `publish_local.sh` that should deal with everything, and I think for the most part our build is not dependent on external dependencies? However, I agree that just running under Rosetta is going to be the best answer here for now. There are some non-trivial changes we'll need to make to our Bazel build to get this working properly.

gcschmit, 2022-12-12:
With the beta 2 release and the validPlatforms updated to include osxuniversal, I'm able to simulate our code. This is using the macOS arm64 image and not the intel image. (I tried installing the intel image, but ran into an issue with the installer.)

PR #28 (2022-12-28): Add macOS Universal Binary support
This PR adds the ability to build a universal binary dylib for the wpilibio component of conduit. Due to various limitations in how our rules work, this was done in a very hacky way where I just copied all the publishing rules specific to macOS, since this universal binary can only be built on systems with full XCode installed, and this rule from `apple_support` behaves strangely on other platforms. Long term we need to write something custom here, and this also has reminded me that we need to look over our `platform_dynamic_library` macro since it is also ki [... cut]
closed 2022-12-28 by camearle20