Mechanical-Advantage/AdvantageKit #26: AutoLog generated files report compiler errors after being built

gcschmit, 2022-12-12:
We are trying the AutoLog annotation with AdvantageKit beta 2. The two files are being generated as expected and building the robot code completes successfully. However, after the build completes successfully, VS Code reports that these two files have compilation errors:

[image]

If I click on each file, the compilation error is immediately resolved as if VS Code realizes that the class is not already defined. This also presents itself when simulating the robot code as the build completes successfully but then, after selecting the simulator tool, VS Code reports that compilation fails. If I continue anyway, everything works as expected.

[image]

We are using spotless, and we've excluded everything under generated-sources.

I'm running macOS 12.6.1 (arm64) and wpilib 2023.1.1-beta-6.

jwbonner, 2022-12-24:
It seems like the simplest solution is to remove the section below from "build.gradle" that registers the generated classes with Gradle and VSCode. This doesn't appear to affect the Gradle build, but it stops VSCode from doing its (incorrect) error checking. The jump-to-definition function in VSCode still seems to still work properly.

```
sourceSets {
    main {
        java {
            srcDirs "src/main/java"
            srcDirs "build/generated/sources/annotationProcessor/java/main"
        }
    }
}
```

I’ve updated the docs and templates in 1115c65

closed 2022-12-24 by jwbonner

camearle20, 2022-12-24:
Note that not putting this in build.gradle in a new project will break VSCode from initially picking up generated sources. When you get compilation errors in VSCode with generated sources the only way I have found to fix it is to do "Clean Java Language Server Workspace" from the command palette. This seems to be an issue with the underlying, Eclipse based language server as this source set configuration shouldn't be necessary to begin with (for example IntelliJ IDEA properly picks up generated sources from a Gradle project with no configuration required).

I'm fine with leaving this out of our docs and seeing what happens, but I suspect this will trip up people who are creating new projects and expecting AutoLog to work properly.

jwbonner, 2022-12-24:
I tried creating a new blank project without the "sourceSets", and VSCode seemed to pick up the generated sources correctly. The generated class appears in this "bin" folder that's only visible when the tab is open, but all of the functionality appears to be intact. Am I missing something?

[image]

camearle20, 2022-12-24:
It's quite possible that they updated the langauge server extension somehow and fixed this. I admittedly haven't actually tested this in a while. If this is working now without the build.gradle change then removing it is definitely the right call. Thanks for testing this!