Mechanical-Advantage/AdvantageKit #30: Issues with XboxController through advantagekit

daisystudent341, 2023-01-14:
When using an XboxController class through advantagekit, we experience a Joystick unplugged error, where the program warns us that it is not registering the buttons on the controller. When using an XboxController through Wpilib instead, it works fine.

This did work on the Beta, so we're not sure what the issue could be.

We are using a Logitech Gamepad F310.

jwbonner, 2023-01-14:
Could you check that your `Robot` class inherits from `LoggedRobot` and not `TimedRobot`?

macfraser, 2023-01-26:
I have seen the same with the some base code we started from the command based skeleton. Running a Rio2

jwbonner, 2023-01-26:
@macfraser Could you confirm that the controller appears correctly in the Driver Station on port 0? If that looks good then it would be helpful if you could send the AdvantageKit log file.

lidor51, 2023-10-19:
Any update on that? Since it's seems like a critical issue.

jwbonner, 2023-10-19:
@daisystudent341 @macfraser If you could confirm whether you fixed this that would be useful. I haven't seen any confirmed report of this issue in a recent version of AdvantageKit, though. The current version enforces that the robot class must inherit from LoggedRobot, which was a common tripping point. It's unfortunately easy to mix up joystick ports too, but that's unrelated to AdvantageKit.

I would expect that if other users experienced this problem, we would have additional reports. We'll look into it if we hear anything, but otherwise I'll plan to close this issue in a couple weeks.

closed 2023-10-30 by jwbonner