Mechanical-Advantage/AdvantageKit #34: 2.1.3 release zip file is password protected

randomstring, 2023-02-16:
I'm getting blocked trying to update to AdvantageKit 2.1.3. 

Trying to download the `conduit-wpilibio-2.1.3-windowsx86-64.zip` file I'm greeted with a username/password prompt.

```
Could not determine the dependencies of task ':extractReleaseNative'.
> Could not resolve all files for configuration ':nativeRelease'.
   > Failed to transform conduit-wpilibio-2.1.3-windowsx86-64.zip (org.littletonrobotics.akit.conduit:conduit-wpilibio:2.1.3) to match attributes {artifactType=gr-directory, org.gradle.status=release}.
      > Could not find conduit-wpilibio-2.1.3-windowsx86-64.zip (org.littletonrobotics.akit.conduit:conduit-wpilibio:2.1.3).
        Searched in the following locations:
            https://maven.pkg.github.com/Mechanical-Advantage/AdvantageKit/org/littletonrobotics/akit/conduit/conduit-wpilibio/2.1.3/conduit-wpilibio-2.1.3-windowsx86-64.zip
```

camearle20, 2023-02-16:
The Windows build is currently failing for some reason related to publishing artifacts, so the artifact it's trying to download actually doesn't exist, so GitHub packages fails over to a 403 for some reason. I can take a closer look at this soon to see if we can get this fixed.

randomstring, 2023-02-21:
Looks like this is resolved with version 2.2

closed 2023-10-27 by jwbonner