Mechanical-Advantage/AdvantageKit #35: CAN usage high until inital enabled

itsmichaelyu, 2023-02-18:
[image]

We are facing really high loop run times when the robot is disabled. I'm thinking it could be something related to AdvantageKit, but it could totally be something on our end as well.

jwbonner, 2023-02-18:
Could you send a log file where this occurs? You may need to put it in a zip file to upload to GitHub.

itsmichaelyu, 2023-02-18:
[Feb18_Rattles.wpilog.zip](https://github.com/Mechanical-Advantage/AdvantageKit/files/10775228/Feb18_Rattles.wpilog.zip)

jwbonner, 2023-02-18:
Based on the log, I don't think that this is an issue with AdvantageKit. Here's a graph with the periodic time for AdvantageKit (blue) and the user code (red). I also confirmed that the console capture is typically under a millisecond.

[image]

itsmichaelyu, 2023-02-18:
Yep, we found the problem. It was something we were calling in disabled periodic.

closed 2023-02-18 by itsmichaelyu