Mechanical-Advantage/AdvantageKit #37: Exported Log from AdvantgeScope can't be Replayed

itsmichaelyu, 2023-03-26:
When I try to replay a log that was exported from AdvantageScope that was recording AdvantageKit data from the robot. I'm getting an error that says the log wasn't produced by advantage kit.

jwbonner, 2023-03-26:
This is (mostly) the intended behavior. When connecting over NT, AdvantageScope doesn't have any data from before it connected. For replay to work correctly, the inputs from _every cycle_ need to be included in the log. AdvantageKit also structures its log files to guarantee that data is cleanly divided between cycles with no overlap in timestamps (AdvantageScope log exports are currently ordered by field, not timestamp). That's why AdvantageKit will only read log files that it directly produced (identified using the "[extra header](https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/doc/datalog.adoc#header)" in the WPILOG file).

The only case where the replay would work is if AdvantageScope connected to NT as soon as the code started (within the first loop cycle). We could look at supporting that but I'm not sure I see a compelling use case for it.

itsmichaelyu, 2023-03-27:
I understand the ordering by timestamp being a current problem. But for example if the robot starts disabled and nothing is moving, is there any data that is lost from not having the initial connection? I do understand if it isn't worth the extra development time though.

jwbonner, 2023-03-27:
The main issue is that the replay isn't deterministic any more. You're right that there _probably_ won't be a difference in behavior, but normally AdvantageKit can _guarantee_ that there won't be a difference. As a simple example, we update our pose estimate using AprilTags even when the robot is disabled. Cutting off a few seconds of that probably won't make a big difference because it will ultimately converge to a similar location, but our filtering algorithm (the same as WPILib) means that the estimates will never perfectly match the real robot. For cases where that difference doesn't matter, I wouldn't be opposed to supporting this with a warning printed in the console. In general, I'd say it's worth the small effort to download the log just so there's no doubt in the accuracy of the replay.

closed 2023-03-27 by itsmichaelyu