Mechanical-Advantage/AdvantageKit #47: RobotController.getBatteryVoltage() does not return a voltage in unit tests

jonathandao0, 2023-10-05:
Sample unit test code:
```
@test
public void testBattery() {
    assertEquals(12, RobotController.getBatteryVoltage());
}
```

This test works on a default WPILib robot project, but fails once the robot project implements AdvantageKit. This only appears to affect unit tests. This seems to be a difference in the underlying code from how WPILib use `PowerJNI.getVinVoltage()` and how AdvantageKit uses `voltageVin()` from `SystemData`.

jonathandao0, 2023-10-12:
So this initially came out of my team trying to update to the Phoenix 6 API, where our swerve code was not behaving properly when deployed to our robot, but appeared to work in simulation. To better help with narrowing down the root cause due to limited physical access to the robot, I wanted to test the new API with WPILib's physics sim to verify that if we set a position/velocity control to the motor, the motor would run the PID loop and arrive at its target setpoint after a few loop cycles without oscillations.

I understand my use case is unique and  that `Logger.getInstance().start()` is an acceptable workaround for `RobotController.getBatteryVoltage()`. However, now I'm running into issues with Phoenix 6, where calling `TalonFXSimState.setSupplyVoltage()` does not update `TalonFXSimState.getMotorVoltage()`, which seems to again be related to how AdvantageKit interacts with the underlying JNI code.

jwbonner, 2023-10-12:
AdvantageKit doesn't interact with other vendordeps, and `getMotorVoltage` directly calls the Phoenix JNI (so it isn't affected by AdvantageKit's Java shims). Looking at the method documentation, `setSupplyVoltage` sets the input voltage of the TalonFX but `getMotorVoltage` returns the output. That suggests `getMotorVoltage` won't update unless you tell the simulated motor to move. Are you seeing different behavior without AdvantageKit?

gcschmit, 2023-10-12:
> So this initially came out of my team trying to update to the Phoenix 6 API, where our swerve code was not behaving properly when deployed to our robot, but appeared to work in simulation. To better [...]

I believe the Phoenix 6 simulation code is working as expected in our code base. Here's a link to where we periodically update the simulation-related components for our swerve: https://github.com/HuskieRobotics/3061-lib/blob/9e66e063c5e46f2d7608ad5c6ecbf8b93f634c51/src/main/java/frc/lib/team3061/swerve/SwerveModuleIOTalonFXPhoenix6.java#L445 (still being tested before we pull it into main).

jonathandao0, 2023-10-13:
@jwbonner Yes, the behavior is different when implementing AdvantageKit. I've broken out the code to a separate repository [here](https://github.com/jonathandao0/Phoenix6SimTest), with the [dev-AdvantageKit](https://github.com/jonathandao0/Phoenix6SimTest/tree/dev-AdvantageKit) branch only containing the minimum changes needed to implement AdvantageKit showing the issue. If this ends up being an issue on CTRE's end, I can contact them about it.

@gcschmit My issue is specifically related to using the Phoenix 6 Sim classes in unit tests. When running the code in simulation, it appears to work fine.

jwbonner, 2023-10-13:
@jonathandao0 The issue doesn't seem to be with Phoenix specifically, the tests are failing because both the simulated voltages and enabled state aren't being updated. As @camearle20 said earlier, these inputs are normally updated by AdvantageKit but the expected periodic functions don't run in a unit test like this. While the periodic methods are package private and called by `LoggedRobot`, you can write a method like this to refresh all of the core AdvantageKit data:

```java
private void refreshAkitData() {
  ConduitApi.getInstance().captureData();
  LoggedDriverStation.getInstance().periodic();
  LoggedSystemStats.getInstance().periodic();
}
```

Calling this after [enabling the robot](https://github.com/jonathandao0/Phoenix6SimTest/blob/dev-AdvantageKit/src/test/java/frc/robot/TestTalonFxMotorSim.java#L127) and [updating the battery voltage](https://github.com/jonathandao0/Phoenix6SimTest/blob/dev-AdvantageKit/src/test/java/frc/robot/TestTalonFxMotorSim.java#L56) fixed the issue for me. I'll try to add a section to the docs about unit testing to clarify this, because it's definitely not intuitive behavior.

jwbonner, 2023-10-27:
There is now a section on the "Common Issues" page about unit testing.

https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/docs/COMMON-ISSUES.md#unit-testing

closed 2023-10-27 by jwbonner