Mechanical-Advantage/AdvantageKit #53: Support SwerveModuleState and SwerveModuleState[] in @AutoLog Classes

gcschmit, 2023-11-27:
It would be convenient if the @AutoLog annotation supported SwerveModuleState and SwerveModuleState[] types. We now have an inputs class for an abstracted Drivetrain subsystem, and including the swerve module states in these inputs would be convenient. We can certainly workaround the absence of this feature if needed and implement the toLog and fromLog methods ourselves.

jwbonner, 2023-11-27:
The only reason this isn’t supported right now is that WPILib 2024.1.1-beta-3 doesn’t have built-in support for struct encoding of those types. That means AdvantageKit needs a special case for it, which only works for outputs (since it was always intended to be a temporary fix). WPILib already merged support for struct encoding of SwerveModuleState, so this will be fixed for the next WPILib/AdvantageKit release. I’ll keep this issue open until then.

jwbonner, 2023-12-29:
This was fixed with AdvantageKit [v3.0.0-beta-6](https://github.com/Mechanical-Advantage/AdvantageKit/releases/tag/v3.0.0-beta-6).

closed 2023-12-29 by jwbonner