Mechanical-Advantage/AdvantageKit #65: AutoLogOutput annotation doesn't log data

AndromedaHelix, 2024-01-18:
When adding the `@AutoLogOutput` to some methods, such as in: 
```
@AutoLogOutput(key = "SwerveStates/Measured")
  private SwerveModuleState[] getModuleStates() {
    SwerveModuleState[] states = new SwerveModuleState[4];
    for (int i = 0; i < 4; i++) {
      states[i] = modules[i].getState();
    }
    return states;
  }
```

The method doesn't get logged automatically. I have to manually call `Logger.recordOutput("SwerveStates/Measured", getModuleStates());` in order to get the same result.

jwbonner, 2024-01-18:
The most likely cause is that one of the conditions specified in the [docs](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/docs/RECORDING-OUTPUTS.md#autologoutput-annotation) hasn't been met. These are:

- The class must be in the same package as `Robot` (like `frc.robot`) or a subpackage (like `frc.robot.subsystems`). You can call `AutoLogOutputManager.addPackage(String)` to add additional allowed packages.
- The object with the annotation must be instantiated within the first loop cycle.
- The object must be accessible by a recursive search of the fields of `Robot`, only including the allowed packages in the chain as described above.

Could you confirm whether all of those conditions appear to be satisfied?

AndromedaHelix, 2024-01-19:
Fixed it! Thanks! I didn't knew the class must be in the same packages as Robot. Just called `AutoLogOuputManager.addPackage(com.lib) ` and it worked. 

However this takes me to another question, if I'm logging a Falcon500's position and velocity through the annotation, will it matter if I run an internal PID loop (in the motorcontroller I mean) rather than on a PIDController class? I'm not sure how the deterministic requirement reacts on internal PID loops.

AndromedaHelix, 2024-01-21:
After reading through code and documentation, I've see there is no issue. I'll close this issue as it is now fixed. 

Thanks Jonah. AdvantageKit works great.

closed 2024-01-21 by AndromedaHelix