Mechanical-Advantage/AdvantageKit #68: Issue with wpilib version parsing on windows due to CRLF differences.

Theinatorinator, 2024-01-22:
In [this commit](https://github.com/Mechanical-Advantage/AdvantageKit/commit/b4f6279fcf170b91e94f1d685e81c1def833aa03) version parsing may have been broken
I think this breaks due to CRLF in windows, on windows.  On windows the line ending is \r\n. Earlier in the parsing, the lines are split out by \n. So on Windows, the array is one character longer than Linux, as the \r is left at the end. 

windows 
 `id "edu.wpi.first.GradleRIO" version "2024.2.1" \r\n`
split string:
`  id , edu.wpi.first.GradleRIO,  version , 2024.2.1, \r `
linux 
 `id "edu.wpi.first.GradleRIO" version "2024.2.1" \n`
split string:
`  id , edu.wpi.first.GradleRIO,  version , 2024.2.1 `

Theinatorinator, 2024-01-22:
Just set a .gitattributes file
`*.gradle eol=lf`
to force the gradle file to have lf and it now works.

jwbonner, 2024-01-22:
Thanks, I’ll push a release to fix this shortly. I switched the latest release to 3.0.0 for now until that’s fixed.