Mechanical-Advantage/AdvantageKit #70: Log every N support

tervay, 2024-02-02:
Currently, we log data every 20ms, as per the standard robot cycle time. However, we log data that is often updated slower than 20ms - for example, motor temperature, which is updated every 100ms - and as such, we often get CAN frame stale warnings (which we intentionally cause).

Since this data is only updating every 5 cycles, it would be nice to not have to log it at all. This is a frequent pattern in [glog](https://github.com/google/glog):

```c++
// every 10ms
LOG_EVERY_T(INFO, 0.01) << "Got a cookie";
// every 10 evaluations
LOG_EVERY_N(INFO, 10) << "Got a cookie";
```

An AdvantageKit equivalent might look like:

```java
Logger.recordOutputEveryN(5, "Subsystem/Foo", calculateFoo());
```

However, this is not lazily evaluated, and `calculateFoo()` may still cause the CAN frame stale warnings. Lazily evaluated support could fix this, with some extra syntactic sugar:

```java
Logger.recordOutputEveryN(5, "Subsystem/Foo", () -> calculateFoo());
```

Under the hood, you could store a `HashMap<String, Integer>`, with example key being `"Subsystem/Foo"`, and check the modulus of the value to check if it should use the new value; if not, use the previously logged value. This way, plots in AdvantageScope are still continuous graphs.

jwbonner, 2024-02-02:
I think this would be a relatively simple addition. Since all of the updates are synchronized to the loop cycle, I'm not sure you would even need a field-specific counter (just an overall loop cycle counter). However, this wouldn't cover inputs that need to be similarly limited (like if you logged temperature as an input rather than an output). Given that our recommended structure doesn't have updateInputs directly use any AdvantageKit APIs, I'm not sure there's much to be done about that though.

tervay, 2024-02-02:
> However, this wouldn't cover inputs that need to be similarly limited (like if you logged temperature as an input rather than an output)

This is true, blanked on that. I wonder if an annotation based log every N for inputs would be possible. I'm not sure how glog implements it under the hood. Shrug. The "best" way here is probably relying on user-code to do it in `updateInputs()`, but might as well brainstorm.

jwbonner, 2024-02-02:
I'm not sure there's a way to make it work with annotations without getting into Lombok-style compiler hacking. One alternative would be adding a utility function like this:

```java
Logger.runEveryN(5, () -> {
  inputs.temperature = motor.getTemperature();
});
```

closed 2024-02-20 by jwbonner

Daniel1464, 2024-10-17:
Wait just as a question for this. does timer.advanceIfElapsed() not exist for this purpose?(or can you not use that for akit)

commit 2563bbbe (2024-02-20): Add `runEveryN` method (closes #70)