Mechanical-Advantage/AdvantageKit #72: Unhandled ConcurrentModificationException in LogTable.clone method

steve-carpenter, 2024-02-16:
It appears that this line is occasionally (1 in 5? times) causing a ConcurrentModificationException on initialization. Robot will crash with fatal error, sometimes restart several times until it works.

Stack trace shows:
`LoggedRobot.startCompetition -> Logger.periodicAfterUser -> LogTable.clone`

I'm _assuming_ that that LogTable object passed into the method is being modified while being cloned- which is causing the issue. Does not appear to be happening in simulation. Code is verbatim from example projects & running the latest release. 

https://github.com/Mechanical-Advantage/AdvantageKit/blob/5a047d936bfb2a3c7c09b26628114cb19d711731/junction/core/src/org/littletonrobotics/junction/LogTable.java#L100

jwbonner, 2024-02-16:
I've never observed this issue on any AdvantageKit example project. Is the code you're running completely unmodified or was it adapted to run on your hardware? The example projects aren't in a state to run on a real robot out of the box. Can you link to the code where you're seeing this happen?

steve-carpenter, 2024-02-16:
We've based our code on this example: https://github.com/Mechanical-Advantage/AdvantageKit/tree/main/example_projects/swerve_drive - by that, I mean- we have same configuration in [Robot.java](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/example_projects/swerve_drive/src/main/java/frc/robot/Robot.java) & use the [AutoLog annotation similarly for our subsystem IO. ](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/example_projects/swerve_drive/src/main/java/frc/robot/subsystems/drive/GyroIO.java)

steve-carpenter, 2024-02-16:
Here is the code snippet from Robot.java
```
 @Override
    public void robotInit() {
        CrashTracker.logRobotInit();
        try {
            Logger.recordMetadata("Team", "2024-Robot"); // Set a metadata value
            switch (getRuntimeType()) {
                case kRoboRIO, kRoboRIO2: {
                    Logger.addDataReceiver(new WPILOGWriter("/home/lvuser/logs/"));
                    Logger.addDataReceiver(new NT4Publisher()); // Publish data to NetworkTables
                    powerDistribution = new PowerDistribution(1, PowerDistribution.ModuleType.kRev);
                    break;

                }
                case kSimulation: {
                    setUseTiming(false); // Run as fast as possible
                    Logger.addDataReceiver(new NT4Publisher()); // Publish data to NetworkTables
                    break;
                }
            }

            Logger.start();

            drivetrain = Drivetrain.getInstance();
            launcher = Launcher.getInstance();
```

jwbonner, 2024-02-16:
Are you using threads anywhere in your code?

steve-carpenter, 2024-02-16:
We're following [254's subsystem manager architecture](https://github.com/Team254/FRC-2023-Public/blob/main/src/main/java/com/team254/frc2023/SubsystemManager.java)- so we technically have threads for [these Enabled/Disabled loops ](https://github.com/Team254/FRC-2023-Public/blob/3856510c06ea67407be9bad08557ddba35b62c71/src/main/java/com/team254/frc2023/SubsystemManager.java#L74). Our Logger.processInputs method calls would be called from within the [Loop class (thread)](https://github.com/Team254/FRC-2023-Public/blob/3856510c06ea67407be9bad08557ddba35b62c71/src/main/java/com/team254/frc2023/SubsystemManager.java#L84)

steve-carpenter, 2024-02-16:
Subsystems don't get initialized until after `Logger.start()` in Robot.java FWIW

jwbonner, 2024-02-16:
This could definitely be caused by calling `Logger` methods (especially `processInputs`) from separate threads. However, the more fundamental issue is that using separate threads for robot logic is not compatible with AdvantageKit since they cannot be replayed in simulation. There are more details in the documentation [here](https://github.com/Mechanical-Advantage/AdvantageKit/blob/main/docs/COMMON-ISSUES.md#multithreading). I would recommend either using a single-threaded architecture for subsystems (like WPILib uses by default), or switching to a non-replay logging framework which will probably be a better fit for your needs (particularly WPILib's [built-in logging](https://docs.wpilib.org/en/stable/docs/software/telemetry/datalog.html), which can be paired with [Monologue](https://github.com/shueja/Monologue/wiki) for annotation support).

jwbonner, 2024-11-01:
The issue seems to be resolved. AdvantageKit's output logging methods are not thread safe, which is called out in the Javadocs.

closed 2024-11-01 by jwbonner