Mechanical-Advantage/AdvantageKit #73: Odometry queues are unlimited in size

PeterJohnson, 2024-02-17:
In the example projects, the deques used are ArrayDeque's, which have unbounded capacity.  This means lines such as this one:

https://github.com/Mechanical-Advantage/AdvantageKit/blob/5a047d936bfb2a3c7c09b26628114cb19d711731/example_projects/advanced_swerve_drive/src/main/java/frc/robot/subsystems/drive/PhoenixOdometryThread.java#L131

Can grow quite large (thousands of timestamps) if the CPU is loaded heavily enough such that the queue doesn't get read and cleared frequently.

closed 2024-02-20 by jwbonner

kevinclark, 2024-02-20:
Hey folks - just ran into this in [our usage of this](https://github.com/HighlanderRobotics/Crescendo/pull/25#discussion_r1496207384). Of jdk-classes I think `ArrayBlockingQueue` makes a lot of sense, but I think it's not the best conceptually. If we're going to drop data because the queue has overflowed I'd personally rather drop the oldest data in favor of the newest. I'm suggesting that 8033 uses [Guava's EvictingQueue](https://guava.dev/releases/23.0/api/docs/com/google/common/collect/EvictingQueue.html) to do that. Just wanted to give a heads up in case Guava is an option for you or you've got access to another ring buffer/circular buffer implementation.

PeterJohnson, 2024-02-20:
WPILib includes CircularBuffer and DoubleCircularBuffer classes that we use for filters etc.  

- https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/src/main/java/edu/wpi/first/util/CircularBuffer.java
- https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/src/main/java/edu/wpi/first/util/DoubleCircularBuffer.java

jwbonner, 2024-02-20:
Given that this is an example project and not part of the AdvantageKit library, my goal for the time being is to minimize the changes required by teams to adopt this. `ArrayBlockingQueue` still implements `Queue` so it's a one-line change, and dropping the newest data is a relatively small issue given that the position will be updated on the subsequent loop cycle. We can look at options that handle this more optimally for next year.

commit a6fc30c0 (2024-02-20): Limit queue capacity for advanced swerve example (fixes #73)