Mechanical-Advantage/AdvantageKit #78: FMSInfo log table not present when actually connected to FMS

mjansen4857, 2024-03-01:
We noticed at our competition this weekend that the FMSInfo table is not present in any of our logs when the robot is connected to a real field. If we tether to the robot, it shows up fine.

jwbonner, 2024-03-01:
The FMSInfo table is a top-level NT table, which is separate from the standard AdvantageKit data flow (the fields under "AdvantageKit" in NT and saved to the log). Therefore it won't be included in any AdvantageKit log files.

mjansen4857, 2024-03-04:
Ah, my mistake. I guess the issue we were seeing might be on the WPILib side then. (Our dashboard depended on data from that table which just wasn't being published when connected to the FMS)

Anyways, is it possible to include this data in the log somehow? It seems useful to have.

jwbonner, 2024-03-05:
AdvantageKit actually does publish that table to NT since it's part of a shimmed class, but I haven't heard of any issues with it not appearing. If you are able to recreate that (and confirm that it is not related to the dashboard) that would be good to know.

All of the equivalent data is logged by AdvantageKit under the "DriverStation" top level key, including robot state, alliance station, event name, match number, etc. AdvantageScope actually uses that data internally for features like automatically selecting bumper color. The shimmed class republishes the data directly to NT under "FMSInfo" in the format expected by driver dashboards, and it's only that form which isn't logged (since it's directly published to NT).

mjansen4857, 2024-03-05:
Yeah the FMSInfo table was not published when actually connected to an FMS. The dashboard used the /FMSInfo/IsRedAlliance entry. It worked fine when connecting to our robot directly, but did not work on the actual field. Making a code change to publish this entry ourselves immediately worked. We also spoke with another team who noticed that this table wasn't being published on the field as well.

jwbonner, 2024-03-05:
The only related issue I've heard of is that the "IsRedAlliance" and "StationNumber" values are incorrect due to a breaking change in the alliance station IDs (we're doing a release very soon to fix this). Can you try running a log from the FMS in replay and checking for the FMSInfo table? You can run replay with real timing (not calling `setUseTiming(false)`) if that's helpful.

jwbonner, 2024-11-01:
The issue seems to be resolved.

closed 2024-11-01 by jwbonner