Mechanical-Advantage/AdvantageKit #81: Logger is not Extensible

BrandonDudek, 2024-03-12:
Some teams may want to extend the Logger class to do custom things with it. But, because if the private constrictor, they are not able to.

Can you please update anything in logger marked as `private` to `protected`.

Here is a PR that does that: https://github.com/Mechanical-Advantage/AdvantageKit/pull/79

jwbonner, 2024-03-12:
What is the use case for changing functionality within `Logger`? In general I would consider that very dangerous and likely to create an invalid state. If there are specific parts of the functionality that needs to be exposed we would consider those changes.

jwbonner, 2024-03-13:
AdvantageKit is already in an unusual situation where using it incorrectly can result in subtly broken behavior which is difficult to trace back to the library. I don't think we want to allow users to cause even more severe issues without a very compelling reason.

BrandonDudek, 2024-03-13:
Seems a bit restrictive. But it is your library. You can control how it can be used. 😄

Would you be more open to just removing the private constructor?

jwbonner, 2024-03-13:
Unless there's an important use case I'm missing, I don't see a reason to make changes. Remember that AdvantageKit, like WPILib, is a library targeted at high school students. Reducing footguns is a critical feature of the library, especially since AdvantageKit replaces parts of WPILib in a way that needs to be done correctly to avoid major issues on the field (issues that are already very difficult for FTAs/CSAs to debug).

AdvantageKit is open source, and users are free to use a fork of the library if desired. But allowing the behavior of core AdvantageKit classes to be overridden by the user is a major support burden both for us and the WPILib/event volunteers trying to keep teams functional on the field.

Again, if there is a use case you're trying to implement which AdvantageKit doesn't support we are more than happy to discuss the relevant changes.