Mechanical-Advantage/AdvantageKit #87: Log Files Truncated

randomstring, 2024-03-16:
We're seeing all of our logs for matches truncated. Often only reporting information for the disable portion before autos start. 

We're setting the log path as directed:

```java
        Logger.addDataReceiver(new WPILOGWriter("/media/sda1"));
        Logger.addDataReceiver(new NT4Publisher());
```

It's as if when the system goes to buffer to the logs it fails to reopen a new file handle? I don't know. 

We think we saw an error about "log buffer overrun" or "consider running clean disk" but it scrolled past pretty quick, and we cannot find it in our logs for the above mentioned truncated logs.

Hints of #36 that I filed. We seem to have bad luck with log files. We are logging a *lot* of data. Example log file attached.

[logs.zip](https://github.com/Mechanical-Advantage/AdvantageKit/files/14624951/logs.zip)

jwbonner, 2024-03-16:
Until you find the original error it's difficult to debug this with any certainty. As a starting point I would recommend switching to a new USB drive.

jwbonner, 2024-03-16:
If you aren't doing so already, I would recommend looking for the error message in the Driver Station log instead of the AdvantageKit log.

randomstring, 2024-03-16:
USB disk swam was the first thing I did. Then I had to wait until after a match to get access to the DS logs. 

I think we found the problem. I'm told we have Logger running in two threads, and I know from reading your posts that Logger is not thread safe. We're looking at that. Any suggestion on how to disable Logging for a whole thread? Can we run a second Logger instance?

`Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:386):
  The startCompetition() method (or methods called by it) should have handled the exception above.
See https://wpilib.org/stacktrace for more information.
The above stacktrace can help determine where the error occurred. Warning at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:379): The robot program quit unexpectedly. This is usually due to a code error.
  at frc.robot.Main.main(Main.java:32)
  at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:453)
  at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:365)
  at org.littletonrobotics.junction.LoggedRobot.startCompetition(LoggedRobot.java:117)
  at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:381)
  at frc.robot.Robot.robotPeriodic(Robot.java:168)
  at edu.wpi.first.wpilibj2.command.CommandScheduler.run(CommandScheduler.java:260)
  at frc.robot.subsystems.swerve.Drivetrain.periodic(Drivetrain.java:191)
  at frc.lib.team2930.LoggerEntry.info(LoggerEntry.java:163)
  at org.littletonrobotics.junction.Logger.processInputs(Logger.java:477)
  at frc.robot.subsystems.swerve.gyro.GyroIOInputsAutoLogged.toLog(GyroIOInputsAutoLogged.java:12)
  at org.littletonrobotics.junction.LogTable.put(LogTable.java:487)
  at org.littletonrobotics.junction.LogTable.put(LogTable.java:373)
Error at org.littletonrobotics.junction.LogTable.put(LogTable.java:373): Unhandled exception: java.lang.NullPointerException: Cannot invoke "edu.wpi.first.util.struct.StructBuffer.write(Object)" because "buffer" is null`

randomstring, 2024-03-16:
We are also seeing:

```txt
DataLog: outgoing buffers exceeded threshold, pausing logging--consider flushing to disk more frequently (smaller period)
```

jwbonner, 2024-03-16:
For the outgoing buffers issue, there is an overload of the WPILOGWriter constructor where you can set the flush period (it defaults to 100ms).

There isn't an easy way to disable the Logger for a single thread (and multiple instances are not supported). You'll have to remove all of the logging calls from one of the threads, which should be easy since the vast majority of code written for an AdvantageKit code structure will be single threaded anyway.

randomstring, 2024-03-16:
We haven't found our issue yet. Testing disabling all the Logging we can in our odometry thread to see if that works.
 
Debugging between matches. @jwbonner I added you to our repo, any advice and/or help is appreciated.  BonneyLake branch is our latest.

I'm advising that we turn off logging to USB drive until we sort it out.

jwbonner, 2024-03-16:
My ability to debug this in detail is limited right now since I'm also at an event. Let me know if you figure out anything else.

randomstring, 2024-03-20:
This is getting weirder. We cleaned up all the Logging from alternative threads at the event and we were still having problems with truncated logs. 

Now we're back in the lab and we cannot reproduce the problem. Same code (didn't even re-deploy), without vision, with vision, running auto, running a practice round via the driver station. I don't see how it can be FMS related and not sure how to try and re-create FMS.

randomstring, 2024-03-21:
OK, much less weird. I thought we were testing our branch from competition, but it turns out someone deployed a change wrapping the Logger calls with `synchronize` debating if we have time to revert and re-run out tests to reproduce the problem.

closed 2024-04-27 by randomstring