Mechanical-Advantage/AdvantageKit #89: Custom timestamps for output logging

jwbonner, 2024-04-08:
It would be useful to be able to log outputs using a timestamp override. Use cases include logging vision data with an offset timestamp (accounting for latency) or logging high-frequency odometry.

Needs to be restricted to outputs only (unless we add a built-in queuing system for inputs).

jwbonner, 2025-03-12:
With this change, it would also be helpful to make the logging methods thread-safe.

Ruthie-FRC, 2026-01-13:
I might just work on a solution for 2026 today/when i have time, i can always tweak when 2027 comes around.

jwbonner, 2026-01-13:
I think one of the intended use cases for this would be to log multiple outputs per loop cycle, like if you have multiple odometry samples per main robot loop. One option would be to somehow nest all of that data as a single `LogValue` that is modified to add new samples (and have it decoded in data receivers), but the challenge is coming up with a way to do that cleanly.

Separately, to support different data types with a feature like this would probably require a large number of additional overloads on `recordOutput` (and likely `LogTable`). Implementing that might only be practical by adding some kind of code generation.

blaze-developer, 2026-01-13:
I feel like as AdvantageKit's list of features and data types grows, the more it would benefit from some sort of code generation that simplifies the amount of code that needs to be written.