Mechanical-Advantage/AdvantageKit #100: DriverStation.getAlliance() value incorrect for a small number of robot loops

sjcbulldog, 2024-10-08:
Until this offseason, we had our own robot code framework.  This offseason, we moved to the Command framework and decided to use AdvantageKit as well.  It has worked out well for the most part and we plan to stick with the Command framework and AdvantageKit for next season.  However, in a recent offseason event, the first since moving to the new code base, we noticed a problem.  We have a target tracker that looks at the alliance color and uses this to extract the pose for the april tag that is at the center of the target.  The target tracker then returns the distance and angle to the desired april tag.  During the periodic call for this subsystem, we check the alliance via DriverStation.getAlliance().  It returns an Optional<Alliance> and as long as it is empty, we ignore the value and defer computing the target april tag location.  Once the optional has a value, we compute the desired target april tag location and cache the value.  The problem is that we were always getting red, even when we were on the blue alliance. This meant we cached the wrong value. So, we grabbed the alliance value, converted to a string, and recorded via Logger.recordOutput().  What we saw was when we were blue, we got red to start with for a few robot loops and then it changed to blue. Don't have the logs in front of me, so don't know how many robot loops were red.  We fixed this in our code by just deferring all of our usage of the getAlliance() value until after the robot was enabled to be sure the value was right.  However, my understanding of how this should work after the change this last season to return an Optional is that once getAlliance() returns an Optional with a value, it should be accurate.  I realize this could be either an AdvantageKit issue or an FMS issue.  Our lead mentor is an FTA so we reported this to the FMS team as well already.

sjcbulldog, 2024-10-08:
From a cursory look, it looks like the value AllianceStationID.Unknown is not being handled in the sendMatchData() method.  It seems like the field can send an alliance ID of zero in the driver station data until the alliance is known.  This is just from a quick look so not completely sure.

jwbonner, 2024-10-10:
When you have access to them, it would be helpful if you could send an AdvantageKit log from a match where you saw this occur. This will allow us to confirm what enum value was being read from the HAL, since AdvantageKit logs the value at a lower level than the `getAlliance` method. `sendMatchData` is only used for publishing the data to NetworkTables (for example, if you use a match info widget on your dashboard) and doesn't affect the values returned to the robot code. The only difference is that it looks like WPILib publishes a different default value to NT than AdvantageKit in the "unknown" case, but that should be unrelated to this problem.

sjcbulldog, 2024-10-10:
Attached is a match where you can see the Alliance Station change from 0 -> 2 (red) -> 5 (blue).  It is also interesting that the DSAttached goes true before FMSAttached goes true.  Our DriverStation was probably set to Red1, but this should not come into play at all when at an event as I understand it.

I could not attach a WPILog file, so the path to the file is https://github.com/errorcodexero/allegro2024cmd/blob/main/ggen/Log_24-10-05_16-33-54_e3.wpilog

Thanks

jwbonner, 2024-10-10:
The Driver Station remembers the alliance station from the last time is was connected to the field, so station 2 (red 2) makes sense because that was your alliance station in the [previous match](https://www.thebluealliance.com/match/2024orgg_sf2m1). My understanding is that it's possible for the Driver Station to connect to the robot via the field network before establishing a connection to the FMS, which is what appears to have happened here. Deferring calls to `getAlliance` until enabled is a good solution, or just skip caching the value and use it directly in the loop (with AdvantageKit, calling `DriverStation.getAlliance()` is just returning an internally cached value anyway).

jwbonner, 2024-10-10:
It sounds like this is specifically an issue with some firmware versions of the new radio. I would expect it to be fixed for the 2025 season.

sjcbulldog, 2024-10-10:
Do you have any more details about the new radio.  This event used the new radios and our lead mentor is a WiFi expert that works for Verizon and is the FTA working with first on testing the new radios.  I want to bring this up with him.  Do you know what firmware versions of the radio exhibited this issue?  Thanks

jwbonner, 2024-10-10:
I don't have more details right now, but FIRST/Vivid is aware of the issue.

sjcbulldog, 2024-10-10:
Thanks, I talked to our mentor and he is aware and involved in this now.

closed 2024-10-11 by sjcbulldog