Mechanical-Advantage/AdvantageKit #101: Huge reach, Google Drive or Dropbox sync support.

blaze-developer, 2024-10-11:
It would be WONDERFUL if we could somehow set up something with the framework that automatically puts log files onto a drive folder to be shared across a team. These could be organized by directories named with github branch names, for example, event_firstchance or just based on the time. Just putting this in to get the idea out there. Right now its hard to distribute log files amongst multiple software students and mentors that all want access as soon as a match is over to debug and test.

It could work as a possible method on setup to add a google drive api key or other type thing.

jwbonner, 2024-10-11:
How would you envision this working given that robots are typically not connected to the internet (particularly when configured for events)?

blaze-developer, 2024-10-11:
Perhaps my mind got ahead of itself. Maybe this kind of feature would be better fit for Advantage*Scope*? I apologize.