Mechanical-Advantage/AdvantageKit #102: Make Synchronized Timestamps Nonintrusive

blaze-developer, 2024-10-15:
Using AdvantageKit, my team attempted to measure loop times using the standard expected behavior of  ```Timer.getFPGATimestamp()```. They expected this data to be accurate to the real system time, and spent hours trying to figure out why the timing of the loops were seemingly instant. After I had showed up to the meeting, I had showed them that AdvantageKit shims that method by default to allow for accurate replay. They were upset about this design choice and believed that libraries shouldnt change something that has  a *standard behavior* that is *expected* to work a certain way.

After this kerfuffle, I think that it would be in the interest of this project to make timestamps not synchronize by default, but require subsystems to use a method such as ```Logger.getSynchronizedTimestamp()``` instead of requiring all teams to use the opposite unless they know how to disable this feature. I believe that by either implementing that or another idea it will reduce  confusion with teams that are used to default WPILib behavior.

If you wished to still include the shim, I believe that you should include a method like the above to allow for both, and make that behavior not the default. An opt-in model and not an opt-out model.

jwbonner, 2024-10-15:
Deterministic timestamps are fundamental to the core purpose of AdvantageKit, which is to enable log replay. We allow users to disable deterministic timestamps using the `Logger.disableDeterministicTimestamps` method, but this will always prevent AdvantageKit from functioning correctly and is intended only as a final resort where no other solutions are effective. Shims are required because the current time is used in many parts of WPILib which would otherwise be nondeterministic (such as `WaitCommand` in the command framework).

We include a [documentation page](https://docs.advantagekit.org/data-flow/deterministic-timestamps) which describes the tradeoffs of deterministic timestamps, including the intended solution to exactly the problem you encountered (calling `Logger.getRealTimestamp`). AdvantageKit is targeted at advanced users who understand its tradeoffs and are looking for functionality not provided by alternative frameworks. I don't think it makes sense to change the default behavior such that the framework cannot function as intended.

blaze-developer, 2024-10-15:
Ah, I understand. It is easier to shim just one method and not shim hundreds of classes to allow for just "better user experience". As this is still something that is easy to overlook, could there be a way to make this more apparent to those using the shimmed method? Perhaps a modification of the javadoc or something to warn users? I noticed the javadoc for RobotController was modified, but most users use ```Timer.getFPGATimestamp```.

jwbonner, 2024-10-15:
It's not just about the convenience of not shimming the timestamp in many places, because most uses of the timestamp in user code also need to be made deterministic. The places where the real timestamp is required are the exception, not the rule.

We don't modify the Javadoc of `Timer` simply because that class isn't shimmed. We're actually moving to remove the shims entirely next year, maintaining the functionality using deeper integration with WPILib. We'll make sure that any functionality affected by AdvantageKit is called out appropriately though.

blaze-developer, 2024-10-15:
👀 sounds interesting, by next year do you mean the 2025 season? If not and there is some time between now and then, please consider shimming that class to add that javadoc modification because it could save people lots of time searching for a problem that they may not know is affected by AdvantageKit, like my team has.

P.S.
AdvantageKit and AdvantageScope are some of the most innovative and interesting projects I have ever seen, and I am  excited for whats to come! Much love <3

jwbonner, 2024-11-01:
Shims will be removed for the first 2025 beta release, which resolves this issue.

closed 2024-11-01 by jwbonner