Mechanical-Advantage/AdvantageKit #106: Allow setting color and style when logging fields

katzuv, 2024-10-19:
This will help not needing to set these manually every time opening AdvantageScope, and will also set a uniform style for all developers. 
Perhaps this feature fits more in AdvantageScope, but I am an avid supporter of code-based configuration 🙂.

jwbonner, 2024-10-20:
I don't think is a feature we would be interested in supporting. Code-based configuration is arguably a good idea for driver dashboards, but AdvantageScope is a debugging tool that is explicitly designed to isolate data structure and visualization (for example, this is why `Field2d` is not supported). Users should be free to visualize data however they need for debugging without making code changes. If it's critical that all developers visualize data in the same way, AdvantageScope allows the tab layout to be exported and imported.

katzuv, 2024-10-21:
> If it's critical that all developers visualize data in the same way, AdvantageScope allows the tab layout to be exported and imported.

Is it possible to make this file path custom so we can check it out to Git? That way we could have a layout for debugging at competition, and we can easily share it between laptops.

jwbonner, 2024-10-21:
You can export the tab layout as a JSON file, which can be saved to any path. Saving it using Git should work without issue.

closed 2024-11-01 by jwbonner