Mechanical-Advantage/AdvantageKit #107: Add a method to log custom messages

katzuv, 2024-10-20:
Perhaps utilizing [`DataLogManager.log()`](https://github.wpilib.org/allwpilib/docs/development/java/edu/wpi/first/wpilibj/DataLogManager.html#log(java.lang.String)).

jwbonner, 2024-10-20:
I'm not sure I understand the use case of this, given that AdvantageKit already captures standard console output that is cross-compatible with other tools. You could utilize the same capture mechanism to log data with custom tags, as suggested [here](https://github.com/Mechanical-Advantage/AdvantageScope/issues/240).

katzuv, 2024-10-21:
Yes you are right, that was an oversight as I don't notice the messages logged by `DataLogManager` can be shown in the console. Sorry!

closed 2024-10-21 by katzuv