Mechanical-Advantage/AdvantageKit #109: Example Swerve Drive Talon

blaze-developer, 2024-10-29:
I think it would be quite helpful if there was an Advanced Swerve Skeleton that included support for onboard feedback and feedforward loops with TalonFX motors. This is essential for many teams in reducing latency, and I noticed that in your team's recent robot code, such onboard feedback loops are already implemented. Having something similar to that as well as with the new cleaned-up swerve code and constants in your recent robot repository would significantly improve the templates for swerve drive, making it easier for other teams to switch to this from large library swerves.

jwbonner, 2024-10-29:
We are working on a number of major improvements to the swerve example projects for the 2025 beta, which already includes this change.

blaze-developer, 2024-10-29:
Nice to hear :)

tbowers7, 2024-11-06:
What is the expected 2025-beta release date?

tbeadle, 2024-11-30:
I, too, am interested in how I can start playing around with the new AdvantageKit & example projects.

tbowers7, 2024-12-04:
Yay!  Thanks!  https://github.com/Mechanical-Advantage/AdvantageKit/releases/tag/v4.0.0-beta-1

closed 2024-12-04 by jwbonner