Mechanical-Advantage/AdvantageKit #111: Adavantage kit not supporting ARM based processor in windows

Emma03L, 2024-11-08:
**Describe the bug**
Advantage kit seems to not support ARM based processors in windows.

**To Reproduce**
Build a gradle project with advanatageKit on windows with an ARM based processor.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (required):**
 - OS: Windows 11
 - Project Information: WPILib version- 2024.3.2, Java version- 17

jwbonner, 2024-11-09:
Are you trying to run code natively or through x86 emulation? WPILib doesn't distribute an installer for Windows on ARM, and I'm not aware of any vendors that distribute native artifacts for the platform. There shouldn't be a problem running WPILib and AdvantageKit in emulation, but I don't think we'll want to support Windows natively until WPILib does in the installer.

Emma03L, 2024-11-09:
I'll run it through emulation, wasn't aware there isn't an ARM installer for windows. Thank you for the response!

closed 2024-11-09 by Emma03L