Mechanical-Advantage/AdvantageKit #118: AdvantageKit.json points to invalid URLs when attempting to download java dependencies

juchong, 2024-11-21:
**Describe the bug**
AdvantageKit.json points to invalid URLs when attempting to download java dependencies.

**To Reproduce**
1. Delete your cache or set up a new AdvantageKit project
2. Attempt to compile the project

**Expected behavior**
The URLs that AdvantageKit.json point to should be valid and the project should compile

**Screenshots**
[image]

**Version (required):**
 - OS: Windows 11
 - Project Information: 
 ```
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Projects\oculus-frc-streamer\robot\build.gradle' line: 92

* What went wrong:
Could not determine the dependencies of task ':jar'.
> Could not resolve all files for configuration ':runtimeClasspath'.
   > Could not find org.littletonrobotics.akit.junction:wpilib-shim:3.2.1.
     Searched in the following locations:
       - file:/C:/Users/juanjchong/releases/maven/development/org/littletonrobotics/akit/junction/wpilib-shim/3.2.1/wpilib-shim-3.2.1.pom
       - https://repo.maven.apache.org/maven2/org/littletonrobotics/akit/junction/wpilib-shim/3.2.1/wpilib-shim-3.2.1.pom
     Required by:
         project :
   > Could not find org.littletonrobotics.akit.junction:junction-core:3.2.1.
     Searched in the following locations:
       - file:/C:/Users/juanjchong/releases/maven/development/org/littletonrobotics/akit/junction/junction-core/3.2.1/junction-core-3.2.1.pom
       - https://repo.maven.apache.org/maven2/org/littletonrobotics/akit/junction/junction-core/3.2.1/junction-core-3.2.1.pom
     Required by:
         project :
   > Could not find org.littletonrobotics.akit.conduit:conduit-api:3.2.1.
     Searched in the following locations:
       - file:/C:/Users/juanjchong/releases/maven/development/org/littletonrobotics/akit/conduit/conduit-api/3.2.1/conduit-api-3.2.1.pom
       - https://repo.maven.apache.org/maven2/org/littletonrobotics/akit/conduit/conduit-api/3.2.1/conduit-api-3.2.1.pom
     Required by:
         project :
```

jwbonner, 2024-11-21:
Gradle searches for dependencies in any accessible repo, including Maven central, the WPILib repo, and vendor repos. This is why AdvantageKit isn’t accessible through the URLs it prints out.

It looks like the actual AdvantageKit maven repo is missing. Check that you followed all of the installation instructions in the docs, including the lines that need to be added to build.gradle.

juchong, 2024-11-21:
Thanks! It looks like I missed that step!

closed 2024-11-21 by juchong