Mechanical-Advantage/AdvantageKit #125: v4.0.0-beta-1 vendorlib not finding java library

tbowers7, 2024-12-04:
**Describe the bug**
AdvantageKit v4.0.0-beta-1 vendorlib file does not include an accurate location for the AK Java library.  Error emitted:
```
   > Could not find org.littletonrobotics.akit:akit-java:4.0.0-beta-1.
     Searched in the following locations:
       - file:/Users/tbowers/wpilib/2025/maven/org/littletonrobotics/akit/akit-java/4.0.0-beta-1/akit-java-4.0.0-beta-1.pom
       - https://repo.maven.apache.org/maven2/org/littletonrobotics/akit/akit-java/4.0.0-beta-1/akit-java-4.0.0-beta-1.pom
     Required by:
         root project :
```

**To Reproduce**
Steps to reproduce the behavior:
1. [Import 2024 project](https://docs.wpilib.org/en/latest/docs/software/vscode-overview/importing-last-years-robot-code.html) into a clean install of WPILib 2025.1.1-beta-2 VSCode, and allow to update.
2. Install the AdvantageKit v4.0.0-beta-1 vendorlib using `>WPILib: Manage Vendor Libraries` and using the link to `AdvantageKit.json` on the GitHub release page (https://github.com/Mechanical-Advantage/AdvantageKit/releases/download/v4.0.0-beta-1/AdvantageKit.json).
3. Attempt to build the code.

**Expected behavior**
Code will build and not emit the error seen in the screenshot.

**Screenshots**
[image]

**Version (required):**
 - MacOS 14.7.1
 - Project Information: 
WPILib Information:
Project Version: 2025.1.1-beta-2
VS Code Version: 1.95.3
WPILib Extension Version: 2025.1.1-beta-2
C++ Extension Version: Not Installed
Java Extension Version: 1.36.2024092708
Java Debug Extension Version: 0.58.2024090204
Java Dependencies Extension Version 0.24.0
Java Version: 17
Java Location: /Users/tbowers/wpilib/2025/jdk
Vendor Libraries:
    AdvantageKit (4.0.0-beta-1)
    WPILib-New-Commands (1.0.0)

**Additional context**
Excited to be trying out the beta of v4.0!

jwbonner, 2024-12-04:
AdvantageKit requires additional setup beyond installing the vendordep. As documented [here](https://docs.advantagekit.org/getting-started/installation/), there are two options for installing AdvantageKit:

* Download one of the AdvantageKit template projects. For importing older projects, you can download the skeleton template and manually copy relevant code to the new project.
* Follow the guide for [existing projects](https://docs.advantagekit.org/getting-started/installation/existing-projects), which walks through all of the project modifications required to support AdvantageKit.

SeanErn, 2024-12-05:
> AdvantageKit requires additional setup beyond installing the vendordep. As documented [here](https://docs.advantagekit.org/getting-started/installation/)

I am having the same issue after following the steps. The actual URL returns a 404 error when querried. Is this intentional?

SeanErn, 2024-12-05:
It looks like the API token got removed or expired in some way. Just used my own API token and it worked fine. Just a heads up!

jwbonner, 2024-12-05:
> The actual URL (https://maven.pkg.github.com/Mechanical-Advantage/AdvantageKit) returns a 404 error when querried. I don't think this is intentional?

This is the expected behavior. The Maven repository only returns a valid response when queried with the fully qualified URL to an artifact, including the authentication information.

> It looks like the API token got removed or expired in some way.

The API token provided in the documentation appears to currently be working.

> I am having the same issue after following the steps.

Can you send a link to your code? I'm still able to build a project after following those instructions. It would also be helpful if you can try building one of the AdvantageKit template projects instead.

SeanErn, 2024-12-08:
Works fine now. Maybe it was just a weird issue with maven at that time.

jwbonner, 2024-12-08:
Great to hear. Feel free to reopen the issue if you encounter it again.

closed 2024-12-08 by jwbonner