Mechanical-Advantage/AdvantageKit #126: Can't run skeleton template on a roborio 1

Emma03L, 2024-12-06:
**Describe the bug**
When trying to run the basic skeleton template on a roborio 1 you get an error saying there is insufficient memory

**To Reproduce**
Open the skeleton template from v4.0.0-beta-1 release.
Build the project and deploy to a roborio 1.

**Expected behavior**
The skeleton template should not cause memory problems.

**Screenshots**
[image]

**Version (required):**
 - OS: Windows 11
 - Wpilib: 2025.1.1-beta-2
 - Java version: 17
 - vendor deps versions the same as the skeleton project

jwbonner, 2024-12-07:
We're experimenting with some different GC options for the AdvantageKit templates this year, with the goal of improving performance on the RIO 1 and 2. After some further testing, there area some specific optimizations that we will probably disable by default because they can cause these types of issues on the RIO 1 (despite improving performance on the RIO 2).

Try updating `build.gradle` to include only the following settings under `frcJava(getArtifactTypeClass('FRCJavaArtifact'))`:

```groovy
frcJava(getArtifactTypeClass('FRCJavaArtifact')) {
    jvmArgs.add("-XX:GCTimeRatio=5")
    jvmArgs.add("-XX:+UseSerialGC")
    jvmArgs.add("-XX:MaxGCPauseMillis=50")
}
```

This is closer to WPILib's default settings, and seems to work well in my testing. We will probably switch the default GC in the templates to this configuration, with documentation for additional options RIO 2 users may want to enable.

21-CameronM, 2024-12-08:
I've been experiencing the same issue on our RIO 1. With the specified changes to the `build.gradle` (which appear to have already been in the `build.gradle` when I downloaded the skeleton template). When SSH'ing into the rio, I'm able to see there's between 30-45MB free, but the program is attempted to allocate ~70MB. 

```
# /tmp/hs_err_pid6134.log
# An error report file with more information is saved as:  
# Native memory allocation (mmap) failed to map 69992448 bytes for committing reserved memory.  
# There is insufficient memory for the Java Runtime Environment to continue. 
```

jwbonner, 2024-12-08:
The updates I posted are not in the posted skeleton template. The lines I posted should be the _only_ configuration under `frcJava`.

jwbonner, 2024-12-26:
This issue should be fixed with the template project updates described above. Feel free to open another issue if you encounter memory issues with the AdvantageKit templates.

closed 2024-12-26 by jwbonner