Mechanical-Advantage/AdvantageKit #131: adding vision and updating pose estimation in talon fx swerve template

varshil1999, 2024-12-19:
I have been trying to integrate the vision template with the talon fx swerve project , but the odometry or the pose estimated robot has never been updated. Please help me for the same

jwbonner, 2024-12-19:
Can you send a code to the code you're working on?

varshil1999, 2024-12-23:
https://github.com/TIS-FRC-2025/AdvantageKit_2025_Beta.git

jwbonner, 2024-12-26:
When running that project in simulation, I'm not seeing an issue with the estimated pose. The odometry updates as expected when driving with a joystick, and vision updates appear to be affecting the pose as well. Note that because the simulated vision is based on the current pose, it shouldn't actually affect the robot position (this is a limitation of vision simulation). To confirm that it's functional, I tried a quick test by offsetting the estimated vision poses to confirm that they are affecting the estimated pose.

If you're still seeing the issue, it would help if you can elaborate on the problem and include a log file as an example (you'll need to put it in a zip archive to upload to GitHub).