Mechanical-Advantage/AdvantageKit #136: BufferOverflowException when logging Pose2d

blaze-developer, 2024-12-20:
When logging a Pose2d from a limelight pose estimation, seemingly every other time we run the robot code, the code crashes on the line logging a Pose2d. It is not null, and the error arises within the struct packing process. This is peculiar because

1. It only seems to happen 70 percent of the time, and other times it works for the entire run cycle.
2. Logging has worked every time if we create a new Pose2d based on the previous one as follows:

```
var loggedPose = new Pose2d(poseToLog.getTranslation(), poseToLog.getRotation());
Logger.recordOutput("pose", loggedPose);
```

```
Error at java.base/java.nio.Buffer.nextPutIndex(Unknown Source): Unhandled exception: java.nio.BufferOverflowException
The startCompetition() method (or methods called by it) should have handled the exception above.
at java.base/java.nio.Buffer.nextPutIndex(Unknown Source)
at java.base/java.nio.DirectByteBuffer.putDouble(Unknown Source)
at edu.wpi.first.math.geometry.struct.Translation2dStruct.pack(Translation2dStruct.java:41)
at edu.wpi.first.math.geometry.struct.Pose2dStruct.pack(Pose2dStruct.java:48)
at edu.wpi.first.math.geometry.struct.Pose2dStruct.pack(Pose2dStruct.java:13)
at edu.wpi.first.util.struct.StructBuffer.write(StructBuffer.java:73)
at org.littletonrobotics.junction.LogTable.put(LogTable.java:520)
at org.littletonrobotics.junction.LogTable.put(LogTable.java:645)
at org.littletonrobotics.junction.Logger.recordOutput(Logger.java:1094)
at frc.robot.subsystems.VisionSubsystem.useSimpleApproach(VisionSubsystem.java:109)
at frc.robot.subsystems.VisionSubsystem.periodic(VisionSubsystem.java:77)
at edu.wpi.first.wpilibj2.command.CommandScheduler.run(CommandScheduler.java:263)
at org.xero1425.base.XeroRobot.robotPeriodic(XeroRobot.java:157)
at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:400)
at org.littletonrobotics.junction.LoggedRobot.startCompetition(LoggedRobot.java:116)
at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:417)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:508)
at frc.robot.Main.main(Main.java:23)
```

Our project is here: https://github.com/errorcodexero/xerofwk2025

jwbonner, 2024-12-20:
That error is typically caused when logging data from separate threads, though the error itself may not be produced from a separate thread. See here: https://docs.advantagekit.org/getting-started/common-issues/multithreading#logging-from-threads

blaze-developer, 2024-12-28:
I see. I was unaware that the codebase I was on had a callback method to our odometry thread which logged an output, moving the logging to a periodic loop, it works as intended. Thanks!

closed 2024-12-28 by blaze-developer