Mechanical-Advantage/AdvantageKit #145: Add hasChanged method to LoggedNetworkNumber  to facilitate updating motor controller gains from smart dashboard

Jon-Zimmerman, 2025-01-02:
**Is your feature request related to a problem? Please describe.**
Using loggedNetworkNumber requires additional logic to facilitate things like motor gains being reapplied to the controller when changed in a dashboard.

**Describe the solution you'd like**
Similar or perhaps identical behavior to the previous loggedTunableNumber class with a hasChanged() function.

**Describe alternatives you've considered**
One option would be to reuse the logic from loggedTunableNumber in each class where the gains need to be reapplied, but this would result in duplicate code and excess verbosity.

**Additional context**
NA

jwbonner, 2025-01-02:
`LoggedTunableNumber` isn't a class that's part of AdvantageKit, and it's something we plan on adding since it's specific to one way of handling tuning for constants. The idea with the `LoggedNetwork` classes to be be minimal wrappers around what would otherwise be calls to the `SmartDashboard` API, with anything higher level handled by the user since there are many potential use cases for network inputs. I'm not sure that `hasChanged` is really universal enough to warrant inclusion in the AdvantageKit classes (for example, 6328's 2024 code makes significant changes to how changes are detected in `LoggedTunableNumber`).

Jon-Zimmerman, 2025-01-02:
My mistake I didn't realize it wasn't part of AdvantageKit previously. That makes sense to keep the wrappers minimal. We'll take inspiration from 6328's 2024 code to solve this problem.

closed 2025-01-02 by jwbonner