Mechanical-Advantage/AdvantageKit #147: Exception when process @AutoLog inputs with record field

RockyXRQ, 2025-01-03:
**Describe the bug**
Keep facing this exception when simulating the robot code
```java
java.lang.IllegalArgumentException: java.lang.ClassCastException@7bd52fb3
	at jdk.internal.reflect.GeneratedMethodAccessor15.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:569)
	at org.littletonrobotics.junction.RecordStruct.lambda$new$10(RecordStruct.java:156)
	at org.littletonrobotics.junction.RecordStruct.pack(RecordStruct.java:327)
	at edu.wpi.first.util.struct.StructBuffer.write(StructBuffer.java:73)
	at org.littletonrobotics.junction.LogTable.put(LogTable.java:520)
	at org.littletonrobotics.junction.LogTable.put(LogTable.java:717)
	at com.nextinnovation.team8214.subsystem.vision.ApriltagVisionIOInputsAutoLogged.toLog(ApriltagVisionIOInputsAutoLogged.java:12)
	at org.littletonrobotics.junction.Logger.processInputs(Logger.java:440)
	at com.nextinnovation.team8214.subsystem.vision.ApriltagVision.updateInputs(ApriltagVision.java:135)
	at com.nextinnovation.team8214.subsystem.vision.ApriltagVision.periodic(ApriltagVision.java:100)
	at com.nextinnovation.team8214.util.VirtualSubsystem.periodicAll(VirtualSubsystem.java:15)
	at com.nextinnovation.team8214.Robot.robotPeriodic(Robot.java:124)
	at edu.wpi.first.wpilibj.IterativeRobotBase.loopFunc(IterativeRobotBase.java:400)
	at org.littletonrobotics.junction.LoggedRobot.startCompetition(LoggedRobot.java:116)
	at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:419)
	at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$1(RobotBase.java:490)
	at java.base/java.lang.Thread.run(Thread.java:840)
```

**To Reproduce**
1. My Inputs class
```java
interface ApriltagVisionIO {
  @AutoLog
  class ApriltagVisionIOInputs {
    boolean connected;

    TargetObservation aimTargetObservation;
    PoseObservation[] poseObservations;
    int[] ids;
  }

  record PoseObservation(double timestamp, Pose3d cameraInField, double ambiguity, int[] ids) {}

  record TargetObservation(Rotation2d yaw, Rotation2d pitch) {}

  default void updateInputs(ApriltagVisionIOInputs inputs) {}
}
```

2. Run the simulator
[image]

3. Generated autolog class
[image]

**Expected behavior**
Running without exception report

**Version (required):**
 - OS: Windows 10
 - Project Information: 
 WPILib Information:
Project Version: 2025.1.1-beta-3
VS Code Version: 1.96.0
WPILib Extension Version: 2025.1.1-beta-3
C++ Extension Version: 1.23.2
Java Extension Version: 1.37.0
Java Debug Extension Version: 0.58.2024090204
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    AdvantageKit (4.0.0-beta-1)
    ChoreoLib (2025.0.0-beta-9)
    CTRE-Phoenix (v5) (5.34.0-beta-4)
    CTRE-Phoenix (v6) (25.0.0-beta-4)
    photonlib (v2025.0.0-beta-8)
    WPILib-New-Commands (1.0.0)

jwbonner, 2025-01-03:
Can you send a link to the full codebase where you see this happening?

RockyXRQ, 2025-01-06:
Hi Jonah for some reason of my team I can't have a public link to u right now, but I add you to our repo, please check this link for the whole project. btw I just created a new project with Advantage Kit 4.0.0 and still have this problem
https://github.com/frcnextinnovation/NI-8583-2024-Preseason

RockyXRQ, 2025-01-06:
Where I see this happening is at https://github.com/frcnextinnovation/NI-8583-2024-Preseason/blob/214b6b563d9fa06c9d442249eb9c12abae219b22/src/main/java/com/nextinnovation/team8214/subsystem/vision/ApriltagVision.java#L135 and https://github.com/frcnextinnovation/NI-8583-2024-Preseason/blob/214b6b563d9fa06c9d442249eb9c12abae219b22/src/main/java/com/nextinnovation/team8214/subsystem/vision/ApriltagVision.java#L138

jwbonner, 2025-01-06:
I'm not able to recreate the issue in sim, but I suspect the problem is that you're logging two records with the same name (`TargetObservation`) and causing an internal conflict. We can add better detection of that case to provide a better warning, but I would start by just renaming one of the records to see if that resolves the issue.

I would also note that several of your record types include invalid fields that AdvantageKit cannot log, which it warns you about in the console. Make sure to only include record fields that use the supported types listed [here](https://docs.advantagekit.org/data-flow/supported-types#records) (which notably does not include arrays).

RockyXRQ, 2025-01-06:
We renamed one of the the record type with different name and it works great now! Thank u so much for the quick response!

closed 2025-01-06 by RockyXRQ