Mechanical-Advantage/AdvantageKit #163: TalonFX 2025 Swerve Template Drives 1/4 Speed

Egg3141592654, 2025-01-25:
**Describe the bug**
While downloading and following the directions to set up the Serve Template from the most recent 4.1.0 release of advantage kit, the observed (and logged) module speeds are off by a factor of approximately 3.9, even after system characterization. Field relative coordinates are working as expected, it's just the drive motors that aren't moving at the correct speed. We verified that the chassis can go faster by pushing the kMaxLinearSpeed to a much higher value (as shown in the screenshot below), but the factor remains consistent.

**To Reproduce**
Steps to reproduce the behavior:
1. Perform steps in [the swerve generator docs](https://docs.advantagekit.org/getting-started/template-projects/talonfx-swerve-template/#setup)
2. Install UBS stick to a Kraken-powered SDSMk4 base robot
3. Drive teleoperated to gather data
4. Advantage Scope to observe mismatch in commanded vs held speed (in m/s) from template logs

**Expected behavior**
The screenshot below shows the error, but with crazy inputs. Ideally these two lines should be near identical, and the robot should be driving faster.

**Screenshots**
[image]

**Version (required):**
 - Advantage Swerve Template: 4.1.0
 - Project Information: WPILib Information:
Project Version: 2025.2.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.1.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    AdvantageKit (4.1.0)
    PathplannerLib (2025.2.1)
    CTRE-Phoenix (v6) (25.2.1)
    Studica (2025.0.1)
    WPILib-New-Commands (1.0.0)

**Additional context**
Add any other context about the problem here.

jwbonner, 2025-01-25:
Did you already complete the tuning steps described [here](https://docs.advantagekit.org/getting-started/template-projects/talonfx-swerve-template#tuning)?

Egg3141592654, 2025-01-26:
While we tuned, we didn't perform the steps exactly as mentioned (using a different sysid characterization, relying on measured dims instead of running the radius routine). We will get on that this week and report back our findings.

Egg3141592654, 2025-02-06:
After a few tries of tuning, you were right and the issue was that we didn't perform the PID/SysID tuning offsets as mentioned in the getting started docs. Closing this out, thanks!

closed 2025-02-06 by Egg3141592654