Mechanical-Advantage/AdvantageKit #164: License incompatibility with proprietary vendor dependencies

ethanc8, 2025-02-08:
The GPL-3.0 forbids combining GPL-3.0 code with proprietary code. Although it's good to encourage teams not to have proprietary code, the GPL-3.0 only requires teams to publish their code if they publish binaries or obfuscated sources for their code. Also, unfortunately many teams have to use proprietary vendordeps such as Phoenix 6, and it's possible WPILib might have proprietary dependencies. So the use of GPL-3.0 is potentially quite problematic. It might be better to switch to the permissive license WPILib is using, or to switch to LGPL-3.0 (which only requires that teams release their modifications to AdvantageKit, rather than the entire codebase), or to provide a GPL exception for linking against proprietary dependencies such as Phoenix 6.

jwbonner, 2025-02-12:
We are actively considering the best way to license AdvantageKit in the future, but we will not be making any changes for the 2025 season. You are always welcome to use alternative solutions for logging if you disagree with the terms of the license.

Our understanding of the GPL-3.0 license is that its full copyleft protections are only applied when distributing _binaries_ that utilize the library. This issue is largely irrelevant since teams don't distribute code binaries; AdvantageKit itself and code derived from its template projects remain GPL licensed, but can exist alongside *source code* that has a non-GPL license and/or uses proprietary libraries (as long as the compiled version is not distributed under a non-GPL license, which is not a scenario that affects FRC teams). Again, we are looking at potential alternatives in the future and welcome you to explore alternative libraries if you disagree with the terms of AdvantageKit's current license.

ethanc8, 2025-02-12:
I believe the GPL still applies to source code distribution as well. However, since the proprietary binaries are not distributed along with the source code, I believe that it probably does not cause copyleft issues.

closed 2025-02-12 by jwbonner