Mechanical-Advantage/AdvantageKit #167: Mismatch in logging between array and non array causing java crash

Kouyang07, 2025-02-17:
**Describe the bug**
If I do
```
Logger.recordOutput("DriveToPose/Setpoints", new Pose2d());
```
Then later I do 
```
Logger.recordOutput("DriveToPose/Setpoint", new Pose2d[] {});
```
Java will crash giving the error EXC_BAD_ACCESS (SIGABRT)

**Expected behavior**
I'm not sure what should happen, perhaps some way to gracefully handle it

**Version (required):**
 - OS: MacOS Sequoia 15.1
 - Project Information: 
WPILib 2025.3.1
Gradle 8.5
Java 17

jwbonner, 2025-02-17:
I'm not able to recreate this issue running in simulation on macOS. Here's the process I'm following for reference;

1. Download the AdvantageKit skeleton project.
2. Add these lines at the end of the `Robot` constructor (I assume "Setpoints" vs "Setpoint" in your code is a typo).

```java
Logger.recordOutput("DriveToPose/Setpoints", new Pose2d());
Logger.recordOutput("DriveToPose/Setpoints", new Pose2d[] {new Pose2d(), new Pose2d()});
```

3. Launch simulation, which runs normally without crashing.

Is there anything else you're doing differently from what I described?

Kouyang07, 2025-02-17:
The only thing I can think of that might have an effect is that I had this inside a command. I had the singular pose2d inside execute while the array of pose2d in the end method. I am on akit version 4.1.1

I came across this while running your [DriveToPose](https://github.com/Mechanical-Advantage/RobotCode2025Public/blob/1acd0d02a7e84f3ba5d88be80e06f66e2cd0a53a/src/main/java/org/littletonrobotics/frc2025/commands/DriveToPose.java)

The exact/specific behavior is attached screenshot

[image]

[image]

jwbonner, 2025-02-18:
I think I tracked down what's happening here. When publishing the fields in that sequence, AdvantageKit ends up publishing the struct field with a value whose size doesn't match the declared type. AdvantageScope handles that case cleanly, but the struct decoder used in some WPILib tools does not. This means that it isn't actually the robot program crashing; it's the simulation GUI. The same thing happens if you connect an NT client like Outline Viewer that uses the same code for decoding the struct.

Theoretically, the WPILib tools shouldn't crash in that case, but the root cause is that AdvantageKit ends up publishing data that doesn't conform to the NT spec. I pushed a fix which should address the problem by printing a warning when a conflicting type like this is logged (and skipping samples published with the incompatible type).

closed 2025-02-18 by jwbonner