Mechanical-Advantage/AdvantageKit #169: Replay Watch Fails But Normal Replay Works

blaze-developer, 2025-02-17:
I am trying out Replay Watch functionality. Normal replay works perfectly fine, but when I run a replay watch, with ``./gradlew replayWatch``, it starts to replay, but fails. It gives no information about what failed or any errors. Is there a way I can configure the mode to print out what errors its facing?

jwbonner, 2025-02-17:
This happens because replay watch is explicitly excluding spotless tasks from running during replay. We can put together a fix at some point, but I can't provide a timeline for that. Essentially it needs to detect whether spotless is installed in the project and change the Gradle command (maybe by calling `./gradlew tasks`, but perhaps there's a cleaner way).

blaze-developer, 2025-02-17:
Could I theoretically just copy the source of ReplayWatch.java and make my own version of it without the exclusions for now?

jwbonner, 2025-02-17:
The `ReplayWatch` class relies on some package-private functions, but could be adapted to work standalone. Alternatively, you can add the spotless plugin to the project and simply disable formatting in the configuration.

blaze-developer, 2025-02-17:
Ah, I'll do that for now.

Baconing, 2025-02-27:
Also jumping in on this, is there any point to run spotless inside of replay watch? 

At worst it could potentially modify code and mismatch the original vs simulated code (albeit this would be improbable at the least, since the bytecode is generally going to be the same). This seems like it should be ran in a Github workflow or pre-commit check, not this gradle task.

If this is required for certain codebases (e.g. Mechanical Advantage's robot) it might be better to setup replayWatch to be able to passthrough these arguments through an environment variable or other method.

The ReplayWatch gradle task isn't very verbose when a task fails, so for most people it takes a minute to pinpoint that issue, and when setting up AdvantageKit I didn't see anything about it.

jwbonner, 2025-02-27:
The issue is that replay watch explicitly _disables_ spotless even when it is not installed, not that it is trying to invoke spotless.

jwbonner, 2025-03-13:
I pushed a fix which allows replay watch to function normally when spotless is not installed, which will be part of the next AdvantageKit release.