Mechanical-Advantage/AdvantageKit #171: Setting the thread priority to 99 for the duration of `robotPeriodic()` can starve important vendor threads if `robotPeriodic()` takes too long

bhall-ctre, 2025-02-18:
At the start of `robotPeriodic()` in many of the examples, the [thread priority is set](https://github.com/Mechanical-Advantage/AdvantageKit/blob/cb7de552c144af211f63e22d76caddeb57cead40/template_projects/sources/talonfx_swerve/src/main/java/frc/robot/Robot.java#L109) to the maximum possible realtime priority of 99 using the FIFO scheduler. For a short-running operation that frequently yields control (such as an odometry thread), this is fine. However, for long-running operations, this risks starving other important threads, since nothing except kernel threads can preempt RT priority 99.

This means that if robot loop timing starts pushing things close to the 20ms max, various issues will start to pop up, since most robot program threads run on just one of the CPU cores. It is also possible for a team to accidentally change the code at the end of `robotPeriodic()` to pass in `true` and remain in a lower RT priority, instead of switching back to non-realtime. We've received a support request from a team that did this, which combined with long loop timings caused a ton of issues.

As an example, in the context of the Phoenix 6 vendordep, the threads that are preempted by RT priority 99 include the enable frame, fetching new CAN frames from the CANivore, and any thread calling `BaseStatusSignal::WaitForAll()`. If `robotPeriodic()` starves these threads for too long, it can cause stale CAN frame warnings, "timestamp out of sync" errors, etc.

jwbonner, 2025-02-19:
I believe the data from our testing showed that the improvement from setting RT priority during the command scheduler was measurable but relatively small, so I think it's reasonable to disable that behavior by default (with a comment and documentation about how/why to enable it). This is a similar approach that we take for some of the more aggressive GC optimizations that improve loop cycle timing. We don't include any documentation suggesting that teams keep RT priority on continuously (not sure why a team would make that change), but we can add a warning about that too. Of course this won't affect teams that are already using the old versions of the template, but the provided implementation shouldn't be a significant concern especially for teams that are already operating well within the 20ms loop cycles.

bhall-ctre, 2025-02-19:
> We don't include any documentation suggesting that teams keep RT priority on continuously (not sure why a team would make that change)

In the context of the support request we got, it was basically a copy-paste error on the user's part. It was just unfortunate for them that this particular copy-paste error is critically bad and breaks a lot of behaviors.

> I think it's reasonable to disable that behavior by default (with a comment and documentation about how/why to enable it)

That seems reasonable to me, I doubt the average team will see any measurable benefits from enabling RT on their main thread.

gcschmit, 2025-02-19:
Sharing just to provide some light on why some team may have code like this. We are a team that that was passing in true at the end of robotPeriodic, although not the team that sent in the support request. We did this based on Mechanical Advantages 2024 code base [which also passes true](https://github.com/Mechanical-Advantage/RobotCode2024/blob/fee73633da2a04a496cb7cf8bb0f6d78a67ae8d1/src/main/java/org/littletonrobotics/frc2024/Robot.java#L330) at the end of robotPeriodic.

jwbonner, 2025-02-19:
While understandable, we do **NOT** recommend using 6328's competition code as reference in this way. The AdvantageKit templates are designed to provide the best experience for a large number of teams; this is never a consideration for 6328's own robot code. We frequently experiment with alternative approaches that carry significant risks, with an understanding of possible side effects and mitigation strategies in place. We always recommend following the examples provided by the AdvantageKit template projects (or ask us about alternatives if you're curious!)

gcschmit, 2025-02-19:
Of course; we absolutely understand the risks. I was just sharing to provide insight into why some teams may have code like that. We have learned a great deal from your code over the years. Thanks for providing that as a resource, along with advising caution! I noticed that this year, at the moment, you are experimenting with keeping the thread marked as real-time with a lower priority. What are you hoping to achieve with this approach?

jwbonner, 2025-02-19:
That's one approach we've been testing as a way to improve the consistency of loop cycle timing. It requires a significant amount of additional work to support reliably, like reducing our reliance on work done in other threads (e.g. NetworkTables) and being very careful about our average loop cycle times to ensure that other threads have time to run. It's basically "high risk high reward", so we're currently focused on doing lots of testing pre-competition to ensure that it is stable enough for our particular requirements.

jwbonner, 2025-02-21:
We've now updated the template projects to disable real-time priority by default, and added a section to the template project documentation explaining how and why users might want to re-enable it.

closed 2025-02-21 by jwbonner