Mechanical-Advantage/AdvantageKit #174: Logger is not thread safe

mjansen4857, 2025-02-28:
**Describe the bug**
Logging with `Logger.recordOutput` from multiple threads can sometimes cause a concurrent modification exception:

```
Error at java.base/java.util.HashMap$HashIterator.nextNode(Unknown Source): Unhandled exception: java.util.ConcurrentModificationException ﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 1 ﻿﻿ The robot program quit unexpectedly. This is usually due to a code error.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information. ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:433) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿﻿﻿ERROR ﻿﻿ 1 ﻿﻿ The startCompetition() method (or methods called by it) should have handled the exception above. ﻿﻿ edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:440) ﻿﻿﻿
﻿﻿﻿﻿﻿﻿ 	at java.base/java.util.HashMap$HashIterator.nextNode(Unknown Source) ﻿
﻿﻿﻿﻿﻿﻿ 	at java.base/java.util.HashMap$EntryIterator.next(Unknown Source) ﻿
﻿﻿﻿﻿﻿﻿ 	at java.base/java.util.HashMap$EntryIterator.next(Unknown Source) ﻿
﻿﻿﻿﻿﻿﻿ 	at java.base/java.util.HashMap.putMapEntries(Unknown Source) ﻿
﻿﻿﻿﻿﻿﻿ 	at java.base/java.util.HashMap.putAll(Unknown Source) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.littletonrobotics.junction.LogTable.clone(LogTable.java:104) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.littletonrobotics.junction.Logger.periodicAfterUser(Logger.java:365) ﻿
﻿﻿﻿﻿﻿﻿ 	at org.littletonrobotics.junction.LoggedRobot.startCompetition(LoggedRobot.java:121) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:419) ﻿
﻿﻿﻿﻿﻿﻿ 	at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:510) ﻿
﻿﻿﻿﻿﻿﻿ 	at frc.robot.Main.main(Main.java:13) ﻿

```

This isn't really a huge deal and can be worked around, but it would be nice to be able to log data from multiple threads, especially since custom CTRE swerve requests are run in a different thread.

**Version (required):**
 - OS: Windows 11
 - Project Information: 
 WPILib Information:
Project Version: 2025.3.1
VS Code Version: 1.96.2
WPILib Extension Version: 2025.3.1
C++ Extension Version: 1.23.2
Java Extension Version: 1.38.0
Java Debug Extension Version: 0.58.1
Java Dependencies Extension Version 0.24.1
Java Version: 17
Java Location: C:\Users\Public\wpilib\2025\jdk
Vendor Libraries:
    AdvantageKit (4.1.1)
    PathplannerLib (2025.2.4)
    CTRE-Phoenix (v6) (25.3.0)
    photonlib (v2025.1.1)
    WPILib-New-Commands (1.0.0)

jwbonner, 2025-02-28:
This is called out in the [documentation](https://docs.advantagekit.org/getting-started/common-issues/multithreading#logging-from-threads) and [Javadocs](https://github.com/Mechanical-Advantage/AdvantageKit/blob/b86d1f82c5e8d4bb8fb6f74c1886c36fd8a93a80/akit/src/main/java/org/littletonrobotics/junction/Logger.java#L430-L432). Since all of the logging is synchronized to the loop cycle it doesn't really make sense to log from threads anyway, though potentially a feature like #89 would change that.

jwbonner, 2025-03-12:
This is currently addressed in the documentation, and I think the feature request is best consolidated under #89. Feel free to reopen if you have additional concerns.

closed 2025-03-12 by jwbonner