Mechanical-Advantage/AdvantageKit #176: Bad DriverStation data causes replay to fail.

blaze-developer, 2025-03-09:
[image]

For some reason, the FMS was giving us bad data for one second at the beginning of some matches.

This gave us a match type of 84 in this instance. This makes a log replay fail due to inaccurate data not being accounted for here:

```
public static void replayFromLog(LogTable table) {
    ...
    DriverStationSim.setMatchType(MatchType.values()[table.get("MatchType", 0)]);
    ...
}
```

84 is not an array index with a length of 4, so code crashes. I haven't seen this bad data on all matches, so I assume it is a problem with the FMS, and not the JNI layer.

jwbonner, 2025-03-09:
We should be able to fix the replay issue without too much trouble. It would also be helpful if you can upload the original WPILOG, DSLog, and DSEvents files so we can take a look at the underlying issue with invalid match data.

blaze-developer, 2025-03-12:
Here:

[Logs.zip](https://github.com/user-attachments/files/19200016/Logs.zip)

jwbonner, 2025-03-31:
AdvantageKit should be automatically renaming the log even if the match number changes multiple times. In the example you gave, is the correct match number ever received according to the AdvantageKit log?

closed 2025-08-12 by blaze-developer