Mechanical-Advantage/AdvantageKit #177: USB Logging causes the canivore to randomly crash

NateDCoder, 2025-03-10:
We are using the Advantage Kit TalonFX template project and were trying to log our robot's data using a USB stick. However, when we plug in the USB stick, it causes the CANivore to crash. The only way to fix this is by either power cycling or rebooting the RoboRIO. There doesn't seem to be any pattern to when it crashes it's happened with both full and dead batteries. Sometimes, we can go an hour without it crashing; other times, it crashes within two minutes.

We checked the Linux terminal on the RIO, but it only showed a general error about the USB port. We removed the USB stick and have since run the robot for over 10+ hours without a single crash. The USB stick was a 4GB USB 2.0 flash drive formatted as FAT32, and when we unplugged it, only 1.6GB of the 4GB was used. Is there anything in the code we need to change to stop this from happening?

CoryNessCTR, 2025-03-10:
Can you elaborate what you mean by the CANivore is crashing? What exactly is happening when you enter the problematic state?

If you can, SSH into the RIO when you're in the problematic state, run `modprobe`, and paste the response here.

Does this happen with other USB Flash Drives? I know many teams that are both logging to USB and running CANivore simultaneously without encountering this issue, so I believe there's something unique to your setup that's causing this.

NateDCoder, 2025-03-13:
Sorry for the late respond, I think it was due to the bad flash drive as I plug in a USB 3.0 and it solved our problem

closed 2025-03-13 by NateDCoder