Mechanical-Advantage/AdvantageKit #179: Confusing Documentation on Power Distribution Logging

Jon-Zimmerman, 2025-03-12:
**Describe the bug**
It is unclear how we are intended to use power distribution logging. One page on the documentation says the values are automatically logged (https://docs.advantagekit.org/data-flow/built-in-logging#power-distribution-data) whereas a different part of the documentation says this line of code is required in robot.java  
`new PowerDistribution(1, ModuleType.kRev); // Enables power distribution logging`
https://docs.advantagekit.org/getting-started/installation/existing-projects.

**To Reproduce**
Steps to reproduce the behavior:
See above links

**Expected behavior**

- These 2 solutions seem incompatible and I would expect only one is the intended way of operation.
- If the LoggedPowerDistribution class is the correct one to use, an example code snippet of using the LoggedPowerDistribution class and better clarification on how it automatically determines the module type and canID would be beneficial. 
- Additional explanation of how to access the power distribution object would be useful to be able to use the switchable channel on the PDH. https://docs.wpilib.org/en/stable/docs/software/can-devices/power-distribution-module.html

**Screenshots**

[image]

[image]

**Version (required):**
Online docs

jwbonner, 2025-03-12:
Power distribution data is logged by default if the power distribution device is available on the CAN bus at the default CAN ID for each vendor (0 for the PDP and 1 for the PDH). This why the template projects do not include any calls to `PowerDistribution` or `LoggedPowerDistribution`.

It looks like the documentation for existing projects was outdated. I've updated it to remove that line, with additional details on the "Built-In Logging" page about the API of `LoggedPowerDistribution` (which is the correct way to configure a custom CAN ID and module type).

closed 2025-03-12 by jwbonner

Jon-Zimmerman, 2025-03-12:
Thanks! Is there anyway for us to access the switchable channel on the pdh then? If not is there a way to disable the automatic logging of the pdh to avoid creating duplicate pdh objects?

jwbonner, 2025-03-12:
The current version of AdvantageKit doesn't touch the built-in `PowerDistribution` class at all, so you can interact with it the same way as any other robot program (i.e. `PowerDistribution` and `LoggedPowerDistribution` don't interact at all).