Mechanical-Advantage/AdvantageKit #189: The Value of Log Replay

blaze-developer, 2025-04-11:
How much value do you get from Log Replay? I see it as a valuable tool, but I have a hard time thinking of many use cases for it. I've only ever run a replay during an event to fix a problem once, and I could have easily verified it without replay. I am curious how much value your team gets out of it. I've seen the examples, however, many of the examples relate to the previous year's problems (like retroreflective tape pose estimation), and I have a hard time finding applications in the recent seasons.

Any recent examples or your thoughts on the matter would be appreciated!
Thanks

jwbonner, 2025-04-11:
6328 has used log replay dozens of times at our events this year to solve problems that would be very difficult or timing consuming to fix any other way. We also use it frequently for offline debugging of issues observed during driver practice/tuning that aren't worth spending valuable robot time to debug. It's particularly impactful for some of the most complex features in our robot code, including auto scoring, superstructure control, and other controls that depend on complex sequencing. We also utilize log replay frequently when providing support to other teams using AdvantageKit, since it allows us to debug and experiment with code without access to a robot (we sometimes provide support to teams _not_ using AdvantageKit, which can be quite frustrating in comparison).

We've given presentations at the FIRST Championship for several years which we update each year with new examples and case studies; the 2025 version will include several new case studies to replace the ones that are becoming outdated, which may provide some ideas for places it might be applicable. One of the challenges with presenting those kinds of "case studies" is that we need to simplify the examples enough that they can be explained in a few minutes. The reality is that most of the issues and testing that we do with log replay relate to niche and/or complex issues that would require an intimate understanding of our robot code to fully appreciate. In general, being able to fix extremely complex issues without access to a robot has been transformation for 6328's ability to quickly iterate on code during the season (even through times when access to the robot is limited).

blaze-developer, 2025-04-11:
Thanks for your thoughtful answer! Ill stop cluttering up your github issues now.

closed 2025-04-11 by blaze-developer